ufrisk/pcileech-fpga #1: Cant Connect To Device

trounce1, 2017-10-30:
Hi,

I have just purchased the SP605 and have flashed the hardware with your pre built binary.

My windows 10 machine when plugged into the SP605 via the FDTI module will not connect to the board.

Error :

PCILEECH : Failed to connect to the device.

Any thoughts?

Thank you

ufrisk, 2017-10-30:
Hi,

Have you downloaded the [http://www.ftdichip.com/Drivers/D3XX.htm](FTD3XX.dll) file from FTDI and placed it in the same folder as the pcileech.exe binary?

trounce1, 2017-10-30:
Yes, i have done this. when i run the pcileech.exe probe command it hangs for a second before returning with the error message.

ufrisk, 2017-10-30:
The FTDI also needs to show up cleanly in Windows Device Manager, and the SP605 have to be connected to PCIe on the target at the time when running the probe or any other command.

How are the status LEDs looking on the SP605? DS3, DS4 should not be lit, while DS5 should be lit if you have a correct PCIe link.

trounce1, 2017-10-30:
Thank you for your fast reply again.

I have checked both device manager and the LED's

I believe they are correct. Attached are photos of both.

Thanks again
[image]
[image]

trounce1, 2017-10-30:
oh its working now. thanks for your help.

closed 2017-10-30 by trounce1

ufrisk, 2017-10-30:
awsome :+1: and thanks for letting me know.

Please let me know if you have any improvements suggestions for the FPGA stuff, I'm currently working on a faster better bitstream, but other than that :)

ufrisk, 2017-10-31:
Thanks,

About the bad TLPs, it's a known bug which I thought I had "resolved" by setting an extra delay in pcileech.exe; It's working on my test hardware. A workaround might be if you set in `device605_601.c` the lines with `usleep(300);` to `usleep(500);` and recompile. Or running pcileech.exe with the option -iosize 0x4000 might also help.

Hopefully these issues will be resolved completely once I release the new bitstream which should resolve these bugs. That is a couple of weeks away though.

trounce1, 2017-10-31:
yeah thats done the trick. speeds back to 70mbs with no bad TLP. i am struggling to get this to work while hotplugging via pcie. still only will connect when booted with spartan 6 plugged in.

ufrisk, 2017-10-31:
Awesome that you got it to work. As previously stated I hope the new bitstream that is planned will resolve these issues.

About hotplug - it's highly dependent on the target system if it will work or not. It's not much I can do about it unfortunately. On some systems it works, on others it doesn't :/