ufrisk/pcileech-fpga #3: Xilinx Artix-35T FPGA

nano2K, 2018-01-27:
Would it be possible to use an evaluation board like this https://www.mouser.com/ds/2/690/arty_rm-1020701.pdf (Xilinx Artix-35T FPGA) with pcileech?  A Spartan 6 is quite expensive for me and my use case unfortunately requires accessing 64 Bit memory. Thank you.

ufrisk, 2018-01-27:
I plan to release support for the [PCIeScreamer](shop.lambdaconcept.com) board in the not too distant future which have an Artix 35T FPGA. I recently had an accident and surgery delaying things quite a bit. Hopefully I'll get it to work and release code in February for this. If you are on a tight budget please wait to get the hardware until I've completely sure I'll be able to pull this off.

As far as the board you linked it's not going to work since it does not have PCI Express - even though the FPGA itself is capable.

I know there are also some other a bit less costly Artix-7 boards that should be able to get to work, namely the Xilinx AC701 (but it's expensive) and various ZedBoard configurations.

nano2K, 2018-01-27:
Sorry to hear about the accident. I hope you get well soon. I think I will wait for an update then, because my experience with FPGAs is still quite limited and I cannot afford "risky" Investments right now. PCIeScreamer looks a lot more reasonable price-wise for me. Anyway thanks for your work and the fast reply.

nano2K, 2018-02-08:
Thank you for your update. I am having trouble flashing the device though:

> $ openocd -f flash_screamer.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
    http://openocd.org/doc/doxygen/bugs.html
none separate
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
adapter speed: 10000 kHz
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
loaded file bscan_spi_xc7a35t.bit to pld device 0 in 0s 219106us
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Error: Unknown flash device (ID 0x00ffffff)

Do you know what I am missing here?

ufrisk, 2018-02-08:
Please try using the `bscan_spi_xc7a35t.bit` from the PCIeScreamer repo `https://github.com/enjoy-digital/pcie_injector`  if you aren't already doing that. Place it alongside the cfg file and try again. Hopefully it will detect another flash memory.

nano2K, 2018-02-08:
Now I am getting:
> Error: Unknown flash device (ID 0x0019ba20)

ufrisk, 2018-02-08:
it's an improvement.

The most recent release of OpenOCD (0.10.0) does not support the SPI flash chip used on the PCIeScreamer it seems like.

The good news is that support is committed into man/latest. I'm sorry but I think you have to download and compile OpenOCD from source if you have this problem.

I had some trouble with dependencies so I flashed my device from Xilinx Vivado with a compatible cable that I had around. But is should work fine with OpenOCD compiled from source with the LambaConcept cable.

ufrisk, 2018-02-08:
Please let me know when/if you have success with this.

nano2K, 2018-02-08:
It works! For some reason my self-compiled version of the current master-branch of OpenOCD did not work (same error as in https://github.com/ufrisk/pcileech-fpga/issues/3#issuecomment-364121532), but I was able to flash the device using the latest binary release of https://github.com/gnu-mcu-eclipse/openocd/releases/tag/v0.10.0-7-20180123

ufrisk, 2018-02-08:
Awesome, huge thanks for the update! I'll update the documentation to reflect this.

closed 2019-10-06 by ufrisk