ufrisk/pcileech-fpga #4: PCIescreamer

Meatballs1, 2018-02-16:
So trying to flash this, using the mega download in the wiki. Seems to work fine. However, power cycling still only shows the red power led. No green leds light up, even when pressing the 'test' button. 

```
plasma pcileech_pciescreamer_3_0_bin # openocd -f flash_screamer.cfg
Open On-Chip Debugger 0.10.0+dev-00299-g6d390e1b-dirty (2018-02-16-15:39)
Licensed under GNU GPL v2
For bug reports, read
    http://openocd.org/doc/doxygen/bugs.html
none separate
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
adapter speed: 10000 kHz
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : Listening on port 3333 for gdb connections
loaded file bscan_spi_xc7a35t.bit to pld device 0 in 0s 220303us
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
flash 'jtagspi' found at 0x00000000
auto erase enabled
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
Info : sector 0 took 118 ms
Info : sector 1 took 115 ms
Info : sector 2 took 116 ms
Info : sector 3 took 116 ms
Info : sector 4 took 117 ms
Info : sector 5 took 115 ms
Info : sector 6 took 116 ms
Info : sector 7 took 115 ms
Info : sector 8 took 117 ms
Info : sector 9 took 115 ms
Info : sector 10 took 116 ms
Info : sector 11 took 116 ms
Info : sector 12 took 114 ms
Info : sector 13 took 115 ms
Info : sector 14 took 116 ms
Info : sector 15 took 118 ms
Info : sector 16 took 117 ms
Info : sector 17 took 114 ms
Info : sector 18 took 114 ms
Info : sector 19 took 115 ms
Info : sector 20 took 115 ms
Info : sector 21 took 119 ms
Info : sector 22 took 113 ms
Info : sector 23 took 115 ms
wrote 1572864 bytes from file pcileech_pciescreamer_top.bin in 13.739050s (111.798 KiB/s)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
read 1507868 bytes from file pcileech_pciescreamer_top.bin and flash bank 0 at offset 0x00000000 in 1.357913s (1084.405 KiB/s)
contents match
shutdown command invoked
```

We have tried two `bscan_spi_xc7a35t.bit` files. The one at https://github.com/enjoy-digital/pcie_injector didn't work for us, the one available from https://github.com/jordens/bscan_spi_bitstreams worked for us. 

```
plasma pcileech_pciescreamer_3_0_bin # sha1sum *.bi*                        
3267adbe1c288a4588b50bdd85025252c28f5eef  bscan_spi_xc7a35t.bit
2115cc2b19d54616871df7f3599d6ebdc38eeb34  pcileech_pciescreamer_top.bin
```

Tried flashing both with a 5v power supply and  power via PCIe.

Meatballs1, 2018-02-16:
Windows identifies the FTDI interface when plugged in over USB but not light blink to indicate USB activity. 

Dip switches set to On/Off/Off

Meatballs1, 2018-02-16:
I guess next step is maybe to try the bscan_spi_xc7a35t.bit from pcie_injector repo and get an openocd version that works with it..

ufrisk, 2018-02-16:
Is LD3 lit after power cycling? LD3 means FPGA is programmed with SPI flash contents and started. If it's not lit FPGA is not programmed.

SW1 will invoke a re-program of FPGA with SPI flash contents, what happens if you press this one?

Also SW2 should be as 1:ON-2:OFF-3:OFF

Meatballs1, 2018-02-16:
I believe LD3 lit up on JTAG flash, but turns off after power cycle, so it looks like it's not being programmed. Will retry with the above files on Monday and see how we get on :)

Meatballs1, 2018-02-19:
Strangely plugged it in today, and the lights are on. LD2 and LD3. We hadn't flashed it over the weekend, and we had plugged it in/out etc. 

No luck connecting to it from toolkit:

```
c:\pcileech\pcileech_files>pcileech.exe dump -v

FPGA: ERROR: Unable to retrieve required Device PCIe ID.
PCILEECH: Failed to connect to the device.
```

Drivers seem installed:

[image]

DLL in folder:

[image]

Meatballs1, 2018-02-19:
A couple of reboots later...

```
c:\pcileech\pcileech_files>pcileech.exe display -vv -pcie_gen1

FPGA: Device Info: PCIeScreamer PCIe gen1 x1 [500,0,1000]

TX: MRd32:  Len: 000 ReqID: 0400 BE_FL: ff Tag: 00 Addr: 00000000
0000    00 00 00 00 04 00 00 ff  00 00 00 00               ............

RX: CplD:   Len: 010 ReqID: 0400 CplID: 0090 Status: 0 BC: 000 Tag: 00 LowAddr: 00
0000    4a 00 00 10 00 90 00 00  04 00 00 00 f3 ee 00 f0   J...............
0010    f3 ee 00 f0 c3 e2 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0020    54 ff 00 f0 bf 18 00 f0  67 18 00 f0 a5 fe 00 f0   T.......g.......
0030    87 e9 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0040    f3 ee 00 f0 57 ef 00 f0  53 ff 00 f0               ....W...S...

RX: CplD:   Len: 010 ReqID: 0400 CplID: 0090 Status: 0 BC: fc0 Tag: 00 LowAddr: 40
0000    4a 00 00 10 00 90 0f c0  04 00 00 40 be 1b 00 c0   J..........@....
0010    4d f8 00 f0 41 f8 00 f0  a0 b0 00 f0 39 e7 00 f0   M...A.......9...
0020    27 a1 00 f0 2e e8 00 f0  8f db 00 f0 00 e0 00 f0   '...............
0030    f2 e6 00 f0 6e fe 00 f0  53 ff 00 f0 53 ff 00 f0   ....n...S...S...
0040    a4 f0 00 f0 c7 ef 00 f0  cb bf 00 c0               ............

RX: CplD:   Len: 010 ReqID: 0400 CplID: 0090 Status: 0 BC: f80 Tag: 00 LowAddr: 00
0000    4a 00 00 10 00 90 0f 80  04 00 00 00 f3 ee 00 f0   J...............
0010    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0020    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0030    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0040    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0               ............

RX: CplD:   Len: 010 ReqID: 0400 CplID: 0090 Status: 0 BC: f40 Tag: 00 LowAddr: 40
0000    4a 00 00 10 00 90 0f 40  04 00 00 40 f3 ee 00 f0   J......@...@....
0010    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0020    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0030    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0040    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0               ............
```

ufrisk, 2018-02-19:
nice :) it seems to be working then, albeit not very stable I guess?

I hope to stabilize this in a future version of the bitstream, but as-is right now I just don't know what I did wrong since almost same code working on AC701 development board which is also Artix-7 FPGA.

Meatballs1, 2018-02-19:
Seems to be fairly reliable from a cold boot. A fair amount of bad PCIe TLP errors but seems to overcome those.

 Ideally would be good to work hot pluggable... Or look at upgrading the hardware :) 

Need to work out some sigs for later ntfs.sys (latest is 10/2/2018).

Meatballs1, 2018-02-19:
Currently only at 830kB/s. I assume because of the errors.

ufrisk, 2018-02-19:
If you get TLP errors it may help adjusting the wait-before-read and wait-after-write timing values; default 500 and 0 uS. You can do this by specifying -device-opt0 and -device-opt1 parameters. Example: `pcileech.exe dump -v -device-opt0 750 -device-opt1 250` This is not guaranteed to work though; and if it works it will impact performance a bit.

PCIe hot plug is always a bit of hit and miss. But as-is the PCIeScreamer is also less reliable than the SP605 and the AC701 solutions until I come up with a solution to the stability issues.

About signatures, you can also try the generic Windows 10 signature; `pcileech.exe kmdload -kmd win10_x64` If you do have trouble with it (blue screens); it's dependant on the system; I can fix the ntfs signature as well. But for memory dumping there is not much need of a kernel implant anymore; but they are very useful for other activities still...

Meatballs1, 2018-02-19:
The read/write delays made no difference except for slowing down the transfer rate. The iosize seems to make a big difference:
```
c:\pcileech\pcileech_files>pcileech.exe dump -iosize 0x1000

 Current Action: Dumping Memory
 Access Mode:    DMA (hardware only)
 Progress:       35 / (unknown)
 Speed:          4 MB/s
 Address:        0x0000000002316000
 Pages read:     8982
 Pages failed:   0
^C
c:\pcileech\pcileech_files>pcileech.exe dump -iosize 0x8000

 Current Action: Dumping Memory
 Access Mode:    DMA (hardware only)
 Progress:       161 / (unknown)
 Speed:          26 MB/s
 Address:        0x000000000A170000
 Pages read:     41328
 Pages failed:   0 

```

ufrisk, 2018-02-19:
awesome :) 0x10000 or 0x18000 is probably even better values in your case

closed 2018-05-15 by ufrisk