ufrisk/pcileech-fpga #5: Virtex 7 VC709

nitinkp470, 2018-03-03:
Hello,
Does PCILeech support VC709 development board?

ufrisk, 2018-03-03:
No.
Currently the SP605 and AC701 among the Xilinx development boards are supported.
It's not likely that I'll add support for the VC709 due to the price tag of the board.

nitinkp470, 2018-03-04:
Thanks for the prompt response. I didn't go through the code yet but I can see you have provided the source codes for all the boards. 
Is there significant changes in the protocol from one board to another or just the pin interface? 
Any suggestions if I try to make the src compatible with VC709.

ufrisk, 2018-03-05:
I don't know, but if you're lucky you could go for the AC701 design, change the PINs and re-generate the IP-cores. The AC701 design works really well for me =)

It seems like the same PCIe core is shared between the different 7-series, but I haven't looked into it in detail. If that is the case I think you might be fine if you are satisfied with PCIe gen2 x2 only. 

To utilize more than 2 lanes you'll have to change the design quite a lot, since the receiving end currently won't be able to handle it. But as PCILeech works now you won't gain any performance out of it anyway so no need really...

You'll need the FT601 FMC addon board as well though.

Please let me know how it goes. If it's just a matter of changing the PINs it would be awesome if we could put the VC709 design out there as well for others to use.

nitinkp470, 2018-03-05:
Sure I will and let you know. Thanks

closed 2019-03-09 by ufrisk