ufrisk/pcileech-fpga #6: PCIeScreamer: ID 0x0666 never detected on target system

PWarn3r, 2018-03-15:
I'm using PCIeScreamer with the pre-built pcileech_pciescreamer.bin binary flashed, however, the PCI device is never detected on my target. I've booted the target into Windows 10 (device manager doesn't list the Xilinx Ethernet Adapter with Device ID 0x0666), also tried booting the target to Linux but lspci doesn't show anything new either when diff'ing. 

The PCIeScreamer is directly inserted into the PCIe 1x slot. Only LD1 is lit on the board:
[image]

Flashing output:
```
user@PWarn3r:~/pcileech_pciescreamer_3_0$ /usr/local/bin/openocd -f flash_screamer.cfg 
Open On-Chip Debugger 0.10.0+dev-g6f700d2 (2018-03-14-17:11)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
adapter speed: 10000 kHz
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : Listening on port 3333 for gdb connections
loaded file bscan_spi_xc7a35t.bit to pld device 0 in 0s 227356us
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
flash 'jtagspi' found at 0x00000000
auto erase enabled
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
Info : sector 0 took 121 ms
Info : sector 1 took 120 ms
Info : sector 2 took 120 ms
Info : sector 3 took 118 ms
Info : sector 4 took 120 ms
Info : sector 5 took 120 ms
Info : sector 6 took 120 ms
Info : sector 7 took 120 ms
Info : sector 8 took 121 ms
Info : sector 9 took 122 ms
Info : sector 10 took 122 ms
Info : sector 11 took 122 ms
Info : sector 12 took 123 ms
Info : sector 13 took 124 ms
Info : sector 14 took 120 ms
Info : sector 15 took 119 ms
Info : sector 16 took 122 ms
Info : sector 17 took 120 ms
Info : sector 18 took 124 ms
Info : sector 19 took 120 ms
Info : sector 20 took 124 ms
Info : sector 21 took 122 ms
Info : sector 22 took 120 ms
Info : sector 23 took 123 ms
wrote 1572864 bytes from file pcileech_pciescreamer_top.bin in 75.291397s (20.401 KiB/s)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
read 1507868 bytes from file pcileech_pciescreamer_top.bin and flash bank 0 at offset 0x00000000 in 1.693188s (869.677 KiB/s)
contents match
shutdown command invoked
```

I've tried:
- verifying dip switches: 1:on, 2:off, 3:off.
- using another PCIe 16x slot
- reflashing the pcileech_pciescreamer_top.bin again
- connecting the USB3 to Linux and running pcileech as root
- connecting the USB3 to Windows 10 with Android USB and FTDI drivers, 64-bit FTD3XX.dll in the same dir

Does anyone have a suggesting where I could look next? Could my device be broken even though I never used it before?

ufrisk, 2018-03-15:
what happens when you press the on-board button marked `SW3`

PWarn3r, 2018-03-15:
Then LD1 lights up; no device shown on the target.

[image]

ufrisk, 2018-03-15:
flashing was successful then.

Sometimes it helps if you power off your target computer and then power on it again.

If it gets a PCIe link LD2 should light up. If LD2 is not lit you won't see your operating system detecting the device yet.

PWarn3r, 2018-03-15:
Thank you! Tried I it on a different motherboard and LD2 indeed did light up. `pcileech.exe probe` also works now. 

So the issue here seems to be my [ASUS P8P67](https://www.asus.com/Motherboards/P8P67_PRO/), I'll reset the BIOS and try different settings to see if that fixes it.

Meatballs1, 2018-03-15:
So we have more reliability in 16x slot, and we have changed the UEFI settings to disable searching for video card in PCIe. That and turning power off completely at PSU when its not being detected...

ufrisk, 2018-03-16:
Maybe a BIOS upgrade will do it as well. I had another case in which we were unable to receive. TLPs before a BIOS upgrade. Configuration went OK with that board though so this seems to be a bit more severe.

Unfortunately it's very hard for me to test on a lot of hardware, it would quickly become way too expensive...

Rainbowrawr, 2018-03-17:
Hey ive got the same problem as you, out of curiosity, do you have an m.2 drive?

xssuser, 2018-07-30:
hi all

When I refreshed it with http://blog.lambdaconcept.com/doku.php?id=products:pcie_screamer

Prompt the following error

[image]

Where is the error

need some helps ,thanks

ufrisk, 2018-07-30:
The OpenOCD that comes with Ubuntu is too old to support the PCIeScreamer.

If you build it from the most recent sources (or possibly download a more recent version) it will support the FPGA and SPI on the PCIeScreamer.

xssuser, 2018-07-30:
i used from http://blog.lambdaconcept.com/doku.php?id=products:jtagserial_programmer#spi_flash_programming

git clone https://github.com/ntfreak/openocd.git
 
cd openocd
 
./bootstrap
./configure
make
 
sudo make install

ufrisk, 2018-07-30:
If you build the main/latest from there it should work. Obviously it doesn't though.

Unfortunately I'm not a very advanced OpenOCD user, it's very hard for me to tell what the error might be. Unfortunately the best advice I can give is to try again :/

(I myself almost always use another programming cable together with xilinx vivado)

ufrisk, 2018-07-31:
Awesome, thanks for letting me know. Good luck with your DMA attacking :)

dom0ng, 2019-01-10:
Hello,
Same issue for me here,
Flashing and building was succeful for me ( default), by the way the PCI-e device is not detected on target PC.
Its on a 16x Pci SLOT, and the LED of my motherboard is ON ( ON when a pci board is put inside ) but not detected by the OS.
The LD2 never light up.
I upgrade my BIOS aswell.

Any suggestions ? Screamer 
Thanks !

closed 2019-12-19 by ufrisk