ufrisk/pcileech-fpga #9: PCIeScreamer flashing Error

tiltmasteR, 2018-07-09:
First of all, thanks for contributing so much to the Community. I ran into a problem while flashing on ubuntu

i use the JTAG-Serial from Lambdaconcept with OPENOCD but it stops working in the jtagspi.cfg with:

`Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Error: 'jtagspi' driver rejected flash bank at 0x00000000; usage: (null)
  flash bank bank_id driver_name base_address size_bytes chip_width_bytes
            bus_width_bytes target [driver_options ...]
  flash banks
  flash list
  mflash bank soc_type base_addr pin_id target`

Can you guide me in the right direction? Everything should be connected correctly and the LD3 is green

Thank you very much!

ufrisk, 2018-07-09:
It was a while since I looked into OpenOCD. I most often use another xilinx programming cable directly from Vivado.

Anyhow, have you followed directions as detailed http://blog.lambdaconcept.com/doku.php?id=products:pcie_screamer

Also, I had problems using the OpenOCD Ubuntu package myself (a couple of months ago so that may have changed). But it worked if I downloaded the most recent version as per the instructions provided by LambdaConcept.

Unfortunately I'm on vacation at the moment so I have a hard time looking into this right now. But please let me know if you get it to work.

tiltmasteR, 2018-07-10:
Nice to hear that,

It's okay, enjoy your vacation ;)
I found out that the ubuntu package has a bug, my openocd works now but now i got several other problems. The connection to the jtag serial board is established tho

```
Info: clock speed 10000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: xc7.tap: IR capture error; saw 0x3f not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
loaded file bscan_spi_xc7a35t.bit to pld device 0 in 0s 219684us
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: xc7.tap: IR capture error; saw 0x3f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unknown flash device (ID 0x00ffffff)

```
i got the last openocd im sure, and i got the last proxybitstream? Serious question, i dont have to plugin the pciescreamer into pcie while flashing?

ufrisk, 2018-07-10:
No need to plug it into PCIe, but it might be a good idea to power it up, either via PCIe or the extra power jack.

tiltmasteR, 2018-07-11:
I broke it down to the errorsource, it is the "init"-command in the flash_screamer.cfg!
i can't get it work, dunno what to do, i recompiled openocd several times (even used apt-get openocd (boardmaker said there is a bug in it tho)). Is there a failure in jtag.cfg maybe?

Help appreciated :)

tiltmasteR, 2018-07-11:
Found the error :)

i HAD to plug it into PCIe too to get enough power to flash it properly ;) Even tho lambdaconcept said i dont have to :( (at least i found the fix after 2 days)

Btw i flashed on a linux ubuntu

ufrisk, 2018-07-11:
Thanks for the update. I'm glad things sorted themselves out. It's good to know that either external power (trough the barrel jack or PCIe) is required. Good luck with your DMA attacking!

I'm closing this thread now since the issue seems to be resolved.

closed 2018-07-11 by ufrisk