ufrisk/pcileech-fpga #10: Me again, Pciescreamer doesn't get detected on target system

tiltmasteR, 2018-07-12:
Me again :(

The flashing was successfull but unfortunately now the Board doesnt get detected on my Windows 10 up2date. I tried IT in my 4x pcie slot and on the picture below i have it in my 8x pcie. Updated my biosversion. Led2 also Lights up, so it should work? Any help would be appreciated

The driver Just doesnt get installed properly (can i download them anywhere?) 

[image]

[image]

ufrisk, 2018-07-12:
That is normal. No driver is required on the target system for this to work.

tiltmasteR, 2018-07-13:
Rip, the i have to dive more into it as it didn't work then. K thanks Just delete this here.

ufrisk, 2018-07-13:
Some times it may work if you cold boot your system; but in your case the target system seems to detect the device so I think it should work.

tiltmasteR, 2018-07-13:
IT works, Even though only after i shutdown my PC. On reboot the Board doesnt get detected tho

ufrisk, 2018-07-13:
Thanks for letting me know. Good luck with everything. The rebooting problem might be because the PCIeScreamer is not always 100% stable.

tiltmasteR, 2018-07-13:
Exactly It stops working once i try to write memory :/ but reading works perfectly fine

closed 2018-08-10 by Asdf144