ufrisk/pcileech-fpga #11: Reading memory of a game through DMA with Pci Screamer

pekiko, 2018-07-15:
How would I do this? I ordered the card so while im waiting for it Im wondering hoe could I do this? Is the Pci Screamer fast enough for that?

ufrisk, 2018-07-15:
Yes, it should be fast enough if the PCIeScreamer is stable for you. If plugging it directly into a PCIe slot it's usually ok.

Read times are around 0.5 ms (if cache disable flag is used) if you access it programatically (need to modify pcileech yourself for this).

If accessing it through the file system speeds vary quite a lot from sub 1ms to 50ms from what I've seen. You should also write "0" into the file <your_process>/.config/cache_file_enable to always get fresh data on every read.

BayleyTheDev, 2018-07-28:
This is a complex matter no? As you need to convert virtual addresses to physical addresses? And also get the base address of the game?

ufrisk, 2018-07-28:
@BayleyTheDev it depends on what you mean with complex. The PCILeech file system already maps virtual memory per-process in a file system for every process. It also already parses things like entry points and presents it in the file system. The file system may however be a bit slow though due to the file system driver I use. Access times vary between 1ms and 50ms from what I've seen. If this is sufficient for your needs all you need to do is to read and write files. The functionality behind it all is quite complex; but it's already in there so no need to change it.

To get below 1ms access speeds consistently you need to interface with the PCILeech code directly. PCILeech is currently not available as a library to interface with so you need to interface with the virtual memory (vmm) subsystem directly - mainly calling vmm.c!VmmReadEx after it's set up.

If there is a need for it (among more than a few people), I may consider creating a library (.DLL) for easy access to the vmm subsystem and other nice to have functions.

As far as games, and other high-end malware, I suspect they may detect the possibility of PCILeech by looking at the hardware device id. If you wish to use PCILeech for the reasons you talk about you may want to look into changing the device ids as well. It's super easy - but you need to download and install xilinx vivado in order to rebuild the FPGA code with different device id.

You may reach me at pcileech@frizk.net for more lengthy questions, or on Twitter for just a short question.

Please note that I will not in any way help in creating cheats for games. I may however help with general purpose functionality that is needed for general memory and process analysis that may be helpful for things like malware analysis.

I recognise that my project may be used to cheat in games - since others obviously already used it for that. I don't have a problem with that. My project is just a tool that can be used in many ways. I however wish to point out and make it absolutely clear that I do not plan to get into the cheat creation business in any way whatsoever.

Flodkozza, 2018-08-03:
Hi I am pretty new to fpga cards but I know C++, how do I actually read the memory? With pcileech or I need to write my own program for that using pcileech? And can I execute c++ code from the fpga card?

ufrisk, 2018-08-03:
@Flodkozza to read virtual memory of a process you can either use the PCILeech Memory Process File System which makes virtual memory of all processes available in a file system. This is a bit slow (1-30ms) depending on random overhead for the file system driver.

Alternatively, you can manually interface with PCILeech (coding required) to read virtual memory. Unfortunately I do not have it as a library (.DLL) so you'll have to hack the code directly... (look into vmm.h, vmmproc.h) This will give you <1ms access speeds.

You cannot run C++ code directly on the FPGA with the current design.

Flodkozza, 2018-08-03:
Its actually good to have random overhead and delay if its not too much because there are scans against inhuman same delay every time. So I can actually read virtual memory if I flash pcileech. Where does the memory dump tho because if I cant run C++ code I cant do anything with that memory reads. I would still need C++ program on my PC?

ufrisk, 2018-08-03:
You'll be able to do analysis on the memory on the PC running PCILeech by just reading. If you wish to execute code on the target you'll have to write memory (modifying in-memory executable code) in the target process in specific ways in order to hijack the execution flow of the target process.

Flodkozza, 2018-08-03:
Is it posible to make a c++ program on the PC that uses the memory reads from pcileech directly? Or can I edit the pcileech code so when it reads the memory it does some if checks on it and then does something. No idea about that programming language but I imagine that should be easy?

ufrisk, 2018-08-03:
Yes, it should be possible. PCILeech is coded in c so it shouldn't be too hard to modify it to integrate it into your project.

I'll try to make a DLL version of PCILeech focused on the virtual memory stuff I have in there in the next few months - which should simplify development a lot. But that is still some time away...

Flodkozza, 2018-08-03:
Im a bit confused about physical and virtual memory. If games have virtual memory(?) But pcileech can read physical memory(again ?) how do I convert one to another and do I do it before dumping or whatever. I usually worked only with ReadProcessMemory in C++ and that shouls be virtual memory.

Flodkozza, 2018-08-03:
Well I still didn't order my PCI Screamer since the customs are expensive here so Im gonna order it to another country when I go there in like 6 months so hopefully the DLL will be done at that time. Also can the Pcileech pc be slow or its gonna affect the speed? My target computer is fine but my other PC where I would run the pcileech from is pretty slow.

closed 2019-03-09 by ufrisk