ufrisk/pcileech-fpga #14: [Question] PCIeScreamer stability

false, 2018-09-12:
I noticed today important stabilities issue with my pciescreamer card.
I already had some problems with it but today it just got worst ; Like failing all the time at retrieving process id or even on initialization ; Losing link after 10 minutes. I needed many reboots and plug/unplug sessions to get it back 'alive'.

My card is connected on a m.2 to pcie adaptator, which is connected on a 12V line from the PSU for the power part.

I am trying to find out how to fix this stability problem.
Does someone know if plugging the external power of the pciescreamer card would help stability in my case ?
Or is that my card or m.2 to adaptator that is defective/not good enough ?
Any other idea ?

I have sometimes used it for hours and hardly had any problem, so I guess something can be done about it ?

ufrisk, 2018-09-12:
The PCIeScreamer stability issues are unfortunately well known and documented. I have not found a solution to it, and neither have anyone else from what I know. If you require stable hardware you have to go for the more expensive xilinx dev boards at the moment. That may change later this year or next year since I'm looking into expanding PCILeech hardware support to more devices.

false, 2018-09-12:
Hello,

Yes I read about it in the README.

I am actually just wondering if the fact it can’t attach any process for 1 hour (after reboots + etc) is indeed a pciescreamer stability issue, or that is not what we call stability issues and my problem would come from something else ?

false, 2018-09-15:
So, if anyone is curious about it : using an external power (12V 2A in my case, old WD external 3.5’ HDD psu) seems to help a lot with my link issues (don’t forget to change the pins) ; I have a good psu in my computer but still. I have only used it for 2 days like that so I will see on the mid/long term how it goes, but so far it’s working a lot better.
I have also changed one thing in my code - a thread that was refreshing every ms, switched to 2ms - but I doubt it’s that change that made so much improvement.

Warning : I don’t say it’s perfect now, sometimes some data just seem to stop being able to be accessed, but it is day and night with my issues of before.

closed 2018-09-15 by false