ufrisk/pcileech-fpga #15: How to spoof/change  Vendor (Manufacturer) ID / Device ID PCIE SCREAMER

finalmouse, 2018-10-04:
Hello,

Somebody can tell me how its possbile to spoorf/change the  Vendor (Manufacturer) ID / Device ID of a pciescreamer?

ufrisk, 2018-10-05:
I hope to put together a short guide how to do it this weekend since more people have been asking of this of late. I have a few other things to look into as well with regards to the FPGA project.

Basically you have to download the Xilinx Vivado, generate the project, then in the project explorer locate the PCIe IP block - double click on it and alter the Device ID's, Classes etc. on the correct tab, save and then compile/synthesisize the project. Some combinations of device settings might not work, and may case problems when booting target computer, but just change to some other values in that case.

ufrisk, 2018-10-19:
I'm closing this issue since it's now described in the advanced build readme.

closed 2018-10-19 by ufrisk