ufrisk/pcileech-fpga #20: Detection vectors pciescreamer on target pc

finalmouse, 2018-10-24:
Hello,

I want to know if there is a scanning on the target computer. Which traces can they find?
Like I already changed the device/vendor id so it looks like a complete other device.

ps: is it possible to make it look like a 1080 gtx (if you have already one to look like a SLI setup

Are there any other traces in the memory of the target they know there is a pciscreamer in the PCIE slot?

Flodkozza, 2018-10-24:
Nah dude ESEA's update is bs. They can't detect it if you know how does this work and how to get rid of all detection vectors. ( ex. if u have aim u need to have an undetected method for sending input, not the standard arduino, not gonna tell you how tho. )

finalmouse, 2018-10-24:
Flodkozza, I know ESEA is just scaring people. My device didnt change from the start, thats why I got banned probably. I dont know about other vectors, I need info about that. I dont use aim no, radar only.
Can we talk in private somwhere?

Thankyou in advance

finalmouse, 2018-10-24:
> 
> 
> Nah dude ESEA's update is bs. They can't detect it if you know how does this work and how to get rid of all detection vectors. ( ex. if u have aim u need to have an undetected method for sending input, not the standard arduino, not gonna tell you how tho. )

Flodkozza, I know ESEA is just scaring people. My device didnt change from the start, thats why I got banned probably. I dont know about other vectors, I need info about that. I dont use aim no, radar only.
Can we talk in private somwhere?

Thankyou in advance

Flodkozza, 2018-10-24:
When did you change your IDs? I think ESEA has silently updated the AC about a week ago to collect data so if you played with original IDs in last 7-10 days, that is probably why you got banned. I also use radar only now because I don't wanna introduce more detection vectors tho. 

Honestly I don't see any other detection vectors besides the IDs if you do it right. Tho Ulf is alot smarter than me so hopefully he will know more 😆 

Tbh all accounts on other sites are the same name as my main account on esea so i wouldn't give that publicly as they do manually ban, so I can't talk privately 😛

finalmouse, 2018-10-24:
> When did you change your IDs? I think ESEA has silently updated the AC about a week ago to collect data so if you played with original IDs in last 7-10 days, that is probably why you got banned. I [...]

Had original device/vendor till 17/10. After that I changed it to a broadcom. But all people with screamer/sp605 who used it few weeks ago till than banwave yesterday are banned.

Alot  of guys say there are more detections vectors when you use pcileech (like localhost) for radar but I cant believe that.

I just want to know how to spoof properly. I think I did but not sure 

[image]

I know who you are and you also know who I am. So we can talk privately where we have no links ;)
We can help eachother out

Flodkozza, 2018-10-24:
Just give me any of your usernames on any site and I'll contact you.

finalmouse, 2018-10-24:
> 
> 
> Just give me any of your usernames on any site and I'll contact you.

add me steam? https://steamcommunity.com/id/5254218412854/

finalmouse, 2018-10-24:
> 
> 
> Just give me any of your usernames on any site and I'll contact you.

talk discord? more private Sheepy#2937

Flodkozza, 2018-10-24:
I don't use discord that often so I'm not sure how to find you xD Try to find me: PakiZaki#0279

I don't use voice chat tho.

finalmouse, 2018-10-24:
> 
> 
> I don't use discord that often so I'm not sure how to find you xD Try to find me: PakiZaki#0279
> 
> I don't use voice chat tho.

accepted, you are offline tho

helloboiss, 2018-11-13:
@Flodkozza you are still save on ESEA/faceit?

closed 2019-03-09 by ufrisk