ufrisk/pcileech-fpga #21: Screamer - Unable to retrieve required Device PCIe ID

IgnitionIWNL, 2018-10-27:
So more or less i was able to successfuly flash, i think?, the xc7a35t.bit, however when im on my target PC i get the infamous "Unable to retrieve required Device PCIe ID [1, v3.3, 0000]" error. ive tried generating my own bin and using the precompiled one from the screamers wiki however no bueno.  Any ideas? Thanks.

IgnitionIWNL, 2018-11-05:
the 3.2 bitstream seemed to work but the next issue i have is that pcileech seems to crash my USB driver if i probe out of memory space.

ufrisk, 2018-11-06:
Is it the USB driver that is crashing or the PCIeScreamer or your target system?

I know probing with the PCIeScreamer is not stable; also some target hardwares just don't like it (= they freeze) especially when being probed at memory mapped devices <4GB. Mbe you could try to probe with `pcileech.exe probe -min 0x100000000` to only probe memory above 4GB?

IgnitionIWNL, 2018-11-07:
It seems to be the USB driver as when it crashes i still show the screamer identifier in device manager on the victim pc. Perhaps im mistaken? I guess it could be the 4GB limit. I will test and repost. Thanks for the help.

closed 2020-01-12 by ufrisk