ufrisk/pcileech-fpga #22: PEB

ch1nw3ll, 2018-11-12:
Hello, 
is it possible to get the PEB Base address for a prccess?

Thanks

ufrisk, 2018-11-12:
The PEB and 32-bit PEB (in 32-bit processes) are used internally but aren't exported by any functions. The addresses are visible in the file system though in each process directory.

I'll release an updated version (as a separate project) that will link into PCILeech, and work even better as compared today some time next week I hope. In that version the PEB and PEB32 will be exported in both the C/C++ API and the new Python API.

ch1nw3ll, 2018-11-18:
Cool, looking forward to seeing it :)

ufrisk, 2018-11-19:
This is now supported in the Memory Process File System - https://github.com/ufrisk/MemProcFS

This will seamlessly link into PCILeech by calling pcileech.dll whenever needed.

I'll remove the existing more limited file system functionality from PCILeech in the next release. PCILeech will focus on DMA attacks and Memory Acquisition and the new project on Memory Analysis.

In reality you'll just include "vmm.dll" instead of "pcileech.dll" in your projects and get the same, or even better, functionality.

closed 2019-03-09 by ufrisk