ufrisk/pcileech-fpga #25: PCIeScreamer flashing error - no debug interface found

CuriosoCat, 2018-12-24:
https://github.com/contactHi, I'm quite new to this and would greatly appreciate any advice or guidance! So I'm facing an error trying to flash the PCIeScreamer card. I'm not sure if it's due to incorrectly following the steps on [http://blog.lambdaconcept.com/doku.php?id=products:pcie_screamer](url) or if it's something wrong with how I connected the hardware. 

The error i'm facing upon executing the command: `openocd -f flash_screamer.cfg` is 

> Error: The specified debug interface was not found (ftdi). 
The following debug interfaces are available: 
1: jlink

**Here is what I've done so far, as per the lambdaconcept blog instructions:**
- Downloaded and unzipped the pcileech_pciescreamer.bin file
- Got the latest version of openocd from github and successfully built it
- Obtained the proxy bit streams from github, extracted bscan_spi_xc7a35t.bit to working folder
- Obtained the flash_screamer.cfg
- Ran the `openocd -f flash_screamer.cfg` and the error occurs

**Here is how I've connected the hardware:**
- JTAG Adapter connected to JTAGSerial Programmer board with the 6 wires jumper cable 
- JTAGSerial Programmer board is attached to the PC via USB
Both of them are connected similar to the picture below, but I'm afraid of pushing the 6 wire jumper cable all the way in, lest it spoils: [image]

- JTAGSerial Programmer board connected to PCIeScreamer card via the JTAG Adapter 
- PCIeScreamer card is inserted and powered via the PCIe slot
- Also, LD1 has no light, LD2 is blinking and LD3 is lit up
- Pressing the test button SW3 has no visible effect

ufrisk, 2018-12-24:
I don't know if this is the error, I'm unfortunately not very knowledgeable about OpenOCD, but I know others had problems with not powering the PCIeScreamer via PCIe or barrel jack. The power that the JTAG flash cable provides alone isn't enough. Power via PCIe should be fine though.

Also make sure you flash the PCILeech bitstream, I don't know if this is the case, but if you flash the LambdaConcept bitstream it will blink the LEDs on the PCIeScreamer.

After a flash the board have to be power cycled for the changes to take effect.

dom0ng, 2019-01-10:
@CuriosoCat  There is a problem with the current version of openocd.
Install a older version of openocd and ur good to go for flashing.

https://github.com/ufrisk/pcileech/issues/75#issuecomment-449143764

closed 2019-11-23 by ufrisk