ufrisk/pcileech-fpga #29: Support for Vivado 2018.3

yaperez-anssi, 2019-01-18:
Hi,

I tried to build the pcileech_fpga firmware for the PCIescreamer board. I've updated my very old Vivado installation to 2018.3 (which is the current version) and it fails to build the project with the following error:

```
ERROR: [BD 5-216] VLNV <xilinx.com:ip:mig_7series:4.1> is not supported for the current part. The latest supported version for this part is:4.2
```

The attached patch fix the issue for me but obviously because of the s/4.1/4.2/ it won't build with older Vivado versions (MIG has been updated to 4.2 in [Vivado 2018.3](https://www.xilinx.com/support/answers/71806.html)). I don't know enough about Vivado to know if there are other/better ways.

I've made a patch (commit d11cf7bb at https://github.com/yaperez-anssi/pcileech-fpga/tree/vivado-2018.3) but I'm not sure it makes sense to submit a pull request. I mainly open the issue to document the problem for other users.

yaperez-anssi, 2019-01-18:
Also note that the build will fail anyway without PR #28

ufrisk, 2019-01-24:
Thank you for your contribution. I missed this issue - but the PR was integrated some time ago.

I'll take a closer look and update the projects to the most recent Vivado version as a whole, and also include the upgrade) after my BlueHat talk later in February.

yaperez-anssi, 2019-01-24:
I guess it's a generic issue in FPGA world, but it seems not really possible to support multiple versions (even minor ones) at once, which is painful for end users.

Do you want me to open a PR then?

ufrisk, 2019-01-24:
Thanks for the offer, but I need to go in there to make changes anyway; I hope I have some news coming up with regards to the fpga parts soon. I'm also going to roll back the MIG changes (to increase stability) and make some additional minor changes.

ufrisk, 2019-03-09:
Closing issues.

I pushed support for 2018.3 a while ago, I also added the upgrade-ip command in it so I hope it will work for a while right now.

Thanks for the help on this one 👍

closed 2019-03-09 by ufrisk