ufrisk/pcileech-fpga #31: AlphaMax NeTV2 compatibility

lokkju, 2019-02-24:
Conceptually, Is there any reason that this couldn't be ported to work with the NeTV2?  https://www.crowdsupply.com/alphamax/netv2

FPGA: Xilinx XC7A35T-2FGG484
On-board RAM: 512 MBytes, 32-bit wide DDR3-800
Video Ports: 2 x HDMI type A inputs, 1 x HDMI type A output, 1 x HDMI type D output. One input/output pair configured for in-line ‘NeTV mode’ video filtering.
Max Video Bandwidth: 1920 x 1080 @ 60 Hz
Developer I/O: PCI-express 2.0 x4 with “hax” GPIO extensions on optional/unused pins, 100Base-T ethernet, micro-USB, microSD, JTAG
Power: Either 12 VDC via barrel jack, or 12 VDC via PCI-express; 10 W max operating power
SBC: Optional Raspberry Pi 3B+ for seamless JTAG configuration and overlay video generation
HDL: Optimized for migen/LiteX Python-based hardware description language
Dimensions: 160 mm x 120 mm x 51 mm (optional plastic case outer dimensions)

ufrisk, 2019-02-24:
Yes, the NeTV2 will be usable for this - but it will require a bit of work.

The FPGA is the same as on the already supported PCIeScreamer and AC701 boards - Artix7.

However the NeTV2 is a bit limited in the ways the "attacker" computer may communicate with it. The HDMI isn't really suitable for this (even though I guess it should be possible in theory). The micro-USB looks like it's UART only - i.e. slow. Remaining is the 100MBit Ethernet. This would give us around 10MB/s over the network. Since there is some overhead I guess we would realistically be looking at dump speeds around 8MB/s unless a compression is added as well.

So yes, it would be possible, very useful, but the speeds wouldn't be great. There will be some work involved. I may look into it myself, but I have so much other things to do chances are that it will take a long time. My things are open source so if you'd be interested to help out I would be happy to accept a PR.

lokkju, 2019-02-24:
My idea was to offload much of the tasks directly to the NeTV2 itself, rather than supporting full memory dumps.  Either way, it seems an interesting hardware addition; I may play with it.  Thanks!

closed 2019-02-24 by lokkju

ufrisk, 2019-10-14:
Support Added for NeTV2.

lokkju, 2019-10-14:
Awesome, and thanks; I'll be playing with this next week, for sure!