ufrisk/pcileech-fpga #32: new PCIe Screamer R02 connected to PCIe extension cable will work OK?

Gentle-Knife, 2019-03-08:
new PCIe Screamer R02 connected to PCIe extension cable will work OK?
Because I not have much money, so I want to verify before I purchase one.
THX!

ufrisk, 2019-03-08:
It works much better than the original one, also it will work ok with an extension cable from what I've seen.

But if stability is paramount for you I still suggest the Xilinx dev boards; the PCIeScreamer R02 (from what I have noticed) is not as good for hot-plugging as the Xilinx dev boards. I have not looked into this much further. As long as the link is detected it works ok though - at least my board.

Also, please understand that I publish a open source software as-is, I'm not involved directly with the PCIeScreamer. I'm not able to give you financial advice on which boards you should get.

Gentle-Knife, 2019-03-11:
What does hot-plugging means?
PCIeScreamer works with ExpressCard(case A)?
or After laptop into sleep mode then replace m.2 wifi chip with PCIeScreamer connect to PCIe m.2 extension cable(case B)? 
I bought one original PCIeScreamer several months ago but in case B it didnot work. And I notice the PCIeScreamer R02 has released. So if the PCIeScreamer R02 works with case B, the world is nice.
Thank you very much!

ufrisk, 2019-03-11:
hot-plugging = inserting the expresscard or powering on the device while the computer is running already - i.e. not having it plugged in when the computer is booting.

yes, it now r02 works with extension cable. I have not tested your configuration though. But for standard PCIe extension cable of "bitcoin mining" model it works for me.

Mera-balou, 2019-03-11:
I've just discovered your repo ufrisk, and i am impressed by your work.
I have just few notions about DMA and i understand that thunderbolt 3 (which deals with PCIe protocol) is now embedded in most of USB-C/3.1 connectors (tell me if i'm wrong).
I have few questions if the DMA attack target is a laptop:
- if i buy PCIe Screamer R02 (which i understand that is it the best value for money, even if the stability isn't perfect), is it possible to plug it into a thunderbolt USB-C/3.1 connectors (of a laptop for example)?
- if so, which connector/dock might be works to plug PCIe Screamer R02 to the USB-C/3.1 laptop connector?
- if not, is it possible to perform DMA attack to a laptop?

ufrisk, 2019-03-11:
Thunderbolt3 is mostly secure by default. Also not all USB-C have Thunderbolt3.

In some circumstances it may be possible to change the default behaviour to insecure/legacy in the BIOS though. But even so, more recent versions of Windows hade additional security around Thunderbolt.

An adapter like [this](https://www.aliexpress.com/item/PCI-Express-PCI-E-to-Thunderbolt-3-to-NGFF-M-key-NVME-AHCI-SSD-Convert-Case/32867546205.html) may be used.

About DMA, it depends, your best should would probably be the Wifi M.2 key A+e card inside the laptop (if you unscrew the back cover) - many laptops have it (but not all) - and then you would need an M.2 key A+E to PCIe adapter.

Also, in both scenarios the PCIeScreamer would have to be powered externally.

Mera-balou, 2019-03-12:
I understand that Thunderbolt3 is mostly secure by default in the bios (i guess that user authorization to approve device before functionality is needed) and fully patched Windows add additional security.
- Is-it possible to bypass theses security with UEFI DMA attack? (load a special kernel module for example)
- If not, and as you told me, my best might be the wifi M.2 key (or ExpressCard for older one i guess). Wich adapter (M.2 key to PCIe and/or Express Card to PCIe)  do you recommend?

Thanks

closed 2020-01-12 by ufrisk