ufrisk/pcileech-fpga #33: PCIE Screamer 100% Fail-Read

Bix3, 2019-07-05:
Hello,

I am trying to get the pciescreamer r02 to work. I flashed it with the precompiled Pcileech software correctly i think, because when i press the control button the led lights up, also it gets successfully detected at the target system. I then connect it to another System, from which I use the command pcileech.exe probe, but its not working. it starts reading but 100% of the Reads are Errors. It looks like this: https://i.imgur.com/N6r3sqf.png.

Things i tried:
- SVM and Iommu disable
- other pci slot
- bios update
- connecting usb to target system itself and reading from there

I hope someone can help me.

ufrisk, 2019-07-07:
1) does the LED LD1 blink when you press button SW3 (i.e. board is correctly flashed).
2) does pcileech show fpga bitstream version when connecting with `pcileech -v -vv display -min 0x1000` (usb connection ok)
3) does pcileech show PCIe ID when connecting with `pcileech -v -vv display -min 0x1000` (last four digits/hex after the version number should be other than 0000) (pcie link ok).

windows have started with some anti-dma mitigations, such as disabling bus mastering on some types of connections for new devices when computer is locked.

Also if not using the board in a standard PCIe slot (expresscard/m.2 and such) it may be a good idea to power the pciescreamer externaly via the barell jack. This is not neccessary when having it powered from a standard PCIe slot.

Also, it sometimes help having the board in there when the computer is booted.

Bix3, 2019-07-07:
Hello, thanks for your answer.

-The LD1 lights up as long as i am holding the button SW3.

-When using the `pcileech -v -vv display -min 0x1000` it shows this: https://i.imgur.com/Rt4jC0x.png

- The PcieScreamer is plugged in standard Pcie Slot, and the power source jumper i switched as described in the manual

- When using the `pcileech -device fpga://1 -v -vv display -min 0x1000` it shows this: `https://i.imgur.com/kJzU3lC.png`

ufrisk, 2019-07-07:
it seems like the screamer is flashed and that you have a pcie connection. Can you after a reboot run:
`pcileech -device fpga://1 -v -vv probe` and see what happens if it's able to read any memory.

You may also add the flag -vvv to see the TLP packets on the wire, it may be useful to see if you get any read completion packets back from the target system.

Bix3, 2019-07-07:
I just tried while booting, its the same as if the pc was on, all pages failed

ufrisk, 2019-07-07:
and with the computer booted and the user logged in?

also you could try to switch pcie slot, I never had that problem myself, but other ppl with similar problems have been reporting success with that in the past.

other than this I'm out of ideas, especially if it's working on another computer.

Bix3, 2019-07-07:
its always pages failed. : / i will try with some other pc, maybe i can get it to work. thank you for the help

gcakir, 2019-07-07:
Did you install pciescreamer into the longer slots (the ones where you install GPUs)?

Bix3, 2019-07-07:
I installed it in pcie x1 (short)

gcakir, 2019-07-07:
I had a similar problem. Install it on the longer ones (x16).

Bix3, 2019-07-09:
i installed it in another slot, and now it dont has Fail reads anymore but after some probing memory it sends me "Bad PCIe TLP received! Should not happen". after this i cant start a probe or something again and only a restart helps.

ufrisk, 2019-07-09:
to adjust packet timing and pcie generation please use
`pcileech.exe -device fpga://1:1000:1000:1000 -v -vv <yourcommand>`
the 1 will force pcie gen1
the 1000:1000:1000 will alter default timeouts in uS to slower (higher) values for read:write:probe - please try your way around here.

Bix3, 2019-07-10:
Hello, this works with the Pcileech and also Memprocfs, but when trying to use the C-Api i get a fail and then all stops working  like before. this is the code:

`#include <Windows.h>
#include <stdio.h>

#include "leechcore.h"
#include "vmmdll.h"
#pragma comment(lib, "vmm")

int main()
{
	bool result = false;
	//Init
	LPSTR test[] = { "", "-device", "fpga:\\1:10000:10000:10000" };
	result = VMMDLL_Initialize(3, test);
	if (true == result) {
		printf("SUCCESS: VMMDLL_Initialize\n");
	}
	else {
		printf("FAIL:    VMMDLL_Initialize\n");		
	}
	//	
	system("pause");
	return 0;
}`

ufrisk, 2019-07-11:
This works for me.
1) Clone MemProcFS repository.
2) Download FTD3XX.dll from ftdichip.com and place it in MemProcFS/files directory
3) Create file test.c in MemProcFS/files directory and fill it with contents as per below
4) From visual studio x64 command prompt:
`cl.exe test.c`
`text.exe`

```C
#include <Windows.h>
#include <stdio.h>
#include "leechcore.h"
#include "vmmdll.h"
#pragma comment(lib, "vmm")

int main(_In_ int argc, _In_ char* argv[])
{
    BOOL result;
    result = VMMDLL_Initialize(3, (LPSTR[]) { "", "-device", "fpga" });
    if(result) {
        printf("SUCCESS: VMMDLL_Initialize\n");
    } else {
        printf("FAIL:    VMMDLL_Initialize\n");
    }
    return 0;
}
```
Please also check out my example in vmm_example solution project.
Also if it's not working change the initializer to:
```C
    result = VMMDLL_Initialize(6, (LPSTR[]) { "", "-device", "fpga://1:1000:1000:1000", "-printf", "-v", "-vv" });
```

Bix3, 2019-07-12:
ok, thank you, i will try that

Bix3, 2019-07-19:
Hello, this works now, thank you

ufrisk, 2019-07-22:
thanks, and good luck with everything :)

closed 2019-07-22 by ufrisk

egssy0, 2022-07-30:
Hi everyone, I'm having the same issue.
What I tried, 

- Switched X1 Slots (I have 2)
- Tried every pcileech parameter, all with the same result "Failed reading memory at the address: 0x*1000"

While buying this card I already made an expense for myself, don't want to change mobo for this :D
Is there any other solution instead of putting it to an x16 Slot? If yes please share with me,
Thanks

haode333, 2022-10-02:
running into similar issue and figured I accidentally disabled screamer in windows device manager. Was thinking it should work anyway but I guess it's not the case.