ufrisk/pcileech-fpga #35: PCIe screamer R02 vs R01 with pcileech

false, 2019-09-02:
Hello,
I have been using your pcileech with a R01, sticking to the v3.2 firmware was for me the best choice as I was suffering performance/stuterring with the 3.4 dram optimized version.
I was wondering what kind of performance improvement I would/could get by moving to a R02 card and the v3.4 ?
With my current installation I don't suffer much of connections errors anymore (still some sometimes, I read the r02 essentially fixes these. Is that like a perfect fix or is it just a bit better than r01 ?

ufrisk, 2019-09-06:
If the R01 is working fine for you and is stable for you the gains will be minimal. A couple of more MB/s and faster when trying to read non-readable memory.

The only difference between the v3.2 and v3.4 bitstreams is an ID inside it which PCILeech detects and for the v3.2 one tries to make more read-retrys on failures and also reads a bit slower to try to partly mitigate stability issues.

false, 2019-09-07:
Ok so no point to update it only for a few mb/s ; May I ask if the combo R02 / 3.4 has an influence on reads latency ? (Or is that also negligible)

ufrisk, 2019-09-08:
It should be 100uS faster in stock PCILeech configuration [R1](https://github.com/ufrisk/LeechCore/blob/master/leechcore/device_fpga.c#L102) vs [R2](https://github.com/ufrisk/LeechCore/blob/master/leechcore/device_fpga.c#L124) - but you may alter those values yourself in PCILeech with the -device command line option as described in the [wiki](https://github.com/ufrisk/LeechCore/wiki/Device_FPGA) for the LeechCore memory acquisition library (used by PCILeech).

So basically, if the R01 is stable for you no need to get the R02 one right now.

false, 2019-09-22:
I have a last question, I still have disconnecting issues with the r01 : sometimes after about an hour I lose connection, and I have to shut down and launch again the target to gain back connection. (I don't experience read or write errors during that hour). Have you ever experienced that behavior and - If yes - does the r02 is the way to fix the issue ?

ufrisk, 2019-09-22:
The R02 is much more stable than the R01 in general yes. Can't answer if it fixes your disconnect problem though.

false, 2019-10-28:
Just for info I switched to the r02 version and it's like day and night! Another thing is I'm working on a MacBook Pro, and switching from the Apple usb-c to usb A to another brand adapter also helped a lot with the ft601 bridge disconnections leading to read/communication errors.

closed 2019-10-28 by false