ufrisk/pcileech-fpga #38: Flashing r02 screamer

b3h3m07h, 2019-11-18:
How do I connect the JTAG flasher to the board? Do I need to solder on pins? The one I received is different then the one in your picture. This is what they sent me. 
[image]
[image]

ufrisk, 2019-11-18:
hmm, if you only received that I understand why there is a problem. If you also received an adapter in the flash kit for 6-pin JTAG to 14-pin JTAG you could stick that in the holes on the PCIeScreamer and then put the lab cable on it. Or if you have a "male" 6x1 lab-cable you could use that as well, just stick it in the holes and it will be enough - no real need for soldering...

but I think you could have received something more in that kit, just buy something to fix it, or contact LambdaConcept.

ufrisk, 2019-11-23:
I'm closing this issue since you seem to have it flashed and this seems to be a LambdaConcept distribution issue.

closed 2019-11-23 by ufrisk