ufrisk/pcileech-fpga #39: How to get the RX: data when using the tllp command

asbridge, 2019-11-22:
Ulf,

I have my R02 PCIeScreamer up and running and I can do memory dumps and get good data so I think things are working properly.

Sorry to be a bother, but I'm having a problem understanding how to receive TLPs and data.

My goal is to be able to send an receive raw TLPs.

I read lots of the information in your forum, but I'm stumped with how to read (and show) the TLPs coming in.  Your "dump" command does it fine, so I know the hardware and SW are capable.

I want to issue (for example) the TLP for a MRd32 exactly like your dump command does below.

I send this command:
 ./pcileech tlp -vvv -vvv -in 000000200e0080ff00000000

There is a pause for 15 seconds then I get just the outbound TX: tlp, no receive. 

TLP: Transmitting PCIe TLP. (use -vvv option for detailed info).

TX: MRd32:  Len: 020 ReqID: 0e00 BE_FL: ff Tag: 80 Addr: 00000000
0000    00 00 00 20 0e 00 80 ff  00 00 00 00               ... ........

 ./pcileech -vvv dump -min 0x0 -max 0x8000 -force -out mem.dmp

TX: MRd32:  Len: 000 ReqID: 0e00 BE_FL: ff Tag: 80 Addr: 00000000
0000    00 00 00 00 0e 00 80 ff  00 00 00 00               ............

TX: MRd32:  Len: 000 ReqID: 0e00 BE_FL: ff Tag: 81 Addr: 00001000
0000    00 00 00 00 0e 00 81 ff  00 00 10 00               ............

TX: MRd32:  Len: 000 ReqID: 0e00 BE_FL: ff Tag: 82 Addr: 00002000
0000    00 00 00 00 0e 00 82 ff  00 00 20 00               .......... .

TX: MRd32:  Len: 000 ReqID: 0e00 BE_FL: ff Tag: 83 Addr: 00003000
0000    00 00 00 00 0e 00 83 ff  00 00 30 00               ..........0.

TX: MRd32:  Len: 000 ReqID: 0e00 BE_FL: ff Tag: 84 Addr: 00004000
0000    00 00 00 00 0e 00 84 ff  00 00 40 00               ..........@.

TX: MRd32:  Len: 000 ReqID: 0e00 BE_FL: ff Tag: 85 Addr: 00005000
0000    00 00 00 00 0e 00 85 ff  00 00 50 00               ..........P.

TX: MRd32:  Len: 000 ReqID: 0e00 BE_FL: ff Tag: 86 Addr: 00006000
0000    00 00 00 00 0e 00 86 ff  00 00 60 00               ..........`.

TX: MRd32:  Len: 000 ReqID: 0e00 BE_FL: ff Tag: 87 Addr: 00007000
0000    00 00 00 00 0e 00 87 ff  00 00 70 00               ..........p.

RX: CplD:   Len: 010 ReqID: 0e00 CplID: 0000 Status: 0 BC: 000 Tag: 80 LowAddr: 00
0000    4a 00 00 10 00 00 00 00  0e 00 80 00 f3 ee 00 f0   J...............
0010    f3 ee 00 f0 c3 e2 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0020    54 ff 00 f0 4d 30 00 f0  f5 2f 00 f0 a5 fe 00 f0   T...M0.../......
0030    87 e9 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0040    f3 ee 00 f0 57 ef 00 f0  53 ff 00 f0               ....W...S...

RX: CplD:   Len: 020 ReqID: 0e00 CplID: 0000 Status: 0 BC: fc0 Tag: 80 LowAddr: 40
0000    4a 00 00 20 00 00 0f c0  0e 00 80 40 60 0b 00 c0   J.. .......@`...
0010    4d f8 00 f0 41 f8 00 f0  30 71 00 f0 39 e7 00 f0   M...A...0q..9...
0020    34 01 00 e8 2e e8 00 f0  d2 ef 00 f0 00 e0 00 f0   4...............
0030    f2 e6 00 f0 6e fe 00 f0  53 ff 00 f0 53 ff 00 f0   ....n...S...S...
0040    a4 f0 00 f0 c7 ef 00 f0  20 35 00 c0 f3 ee 00 f0   ........ 5......
0050    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0060    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0070    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0080    f3 ee 00 f0 f3 ee 00 f0  f3 ee 00 f0               ............

ufrisk, 2019-11-23:
Thanks. This is a bug. I'll fix it tomorrow.

ufrisk, 2019-11-23:
This should be updated now. New sources and updated binary package is available.

Thank you for reporting this and it's nice to see that the R02 is working fine for you. Good luck with your future DMA research and please let me know if you find more issues 👍 

Closing this issue now.

asbridge, 2019-11-25:
Ulf,

Works!  Thanks.  Log below.  One more question.  Why is there a 15 second delay between issuing the TLP command and getting the reply?  (see time command below).

Vince

 time ./pcileech tlp -vvv -vvv -in 000000200e0080ff00000000

TLP: Transmitting PCIe TLP. (use -vvv option for detailed info).

TX: MRd32:  Len: 020 ReqID: 0e00 BE_FL: ff Tag: 80 Addr: 00000000
0000    00 00 00 20 0e 00 80 ff  00 00 00 00               ... ........

RX: CplD:   Len: 020 ReqID: 0e00 CplID: 0000 Status: 0 BC: 080 Tag: 80 LowAddr: 00
0000    4a 00 00 20 00 00 00 80  0e 00 80 00 f3 ee 00 f0   J.. ............
0010    f3 ee 00 f0 c3 e2 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0020    54 ff 00 f0 4d 30 00 f0  f5 2f 00 f0 a5 fe 00 f0   T...M0.../......
0030    87 e9 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0040    f3 ee 00 f0 57 ef 00 f0  53 ff 00 f0 60 0b 00 c0   ....W...S...`...
0050    4d f8 00 f0 41 f8 00 f0  30 71 00 f0 39 e7 00 f0   M...A...0q..9...
0060    34 01 00 e8 2e e8 00 f0  d2 ef 00 f0 00 e0 00 f0   4...............
0070    f2 e6 00 f0 6e fe 00 f0  53 ff 00 f0 53 ff 00 f0   ....n...S...S...
0080    a4 f0 00 f0 c7 ef 00 f0  20 35 00 c0               ........ 5..

real    0m14.902s
user    0m0.377s
sys     0m1.504s

ufrisk, 2019-11-25:
Can you please check with the pcileech fixes I made for the segfault issue. I also included some performance optimizations in which I removed auto-detection of max memory when transmitting TLPs (there is no need for it).

I suspect that the "fix" won't help with this issue, which I'm unable to replicate in my main dev environment. If the issue persists I'll have to do a more thorough investigation in the weekend.

Can you please check if the error persists, if so, does the main wait time happen before or after the TLP is sent / shown on the screen?

asbridge, 2019-11-25:
Ulf,
 
Will try new code late tonight (after family goes to bed...) or early in the
AM.
 
Thanks for the support.
 
Vince

  _____  

tllp command (#39)

<https://github.com/ufrisk/pcileech-fpga/issues/39?email_source=notification
s&email_token=ALANRPGQ6UTADPZ4WXE5OMTQVQ66DA5CNFSM4JQWUQ5KYY3PNVWWK3TUL52HS4
DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEFD3ACI#issuecomment-558346249> it
Q66DANCNFSM4JQWUQ5A> .
<https://github.com/notifications/beacon/ALANRPGHVQV72COBQFXQVCDQVQ66DA5CNFS
M4JQWUQ5KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEFD3ACI
.gif>

asbridge, 2019-11-26:
Ulf,

Fixed!

Now completes in .383 seconds.

Double free also fixed.

Thanks!
Vince

 time ./pcileech tlp -vvv -in 000000200e0080ff00000000

TLP: Transmitting PCIe TLP. (use -vvv option for detailed info).

TX: MRd32:  Len: 020 ReqID: 0e00 BE_FL: ff Tag: 80 Addr: 00000000
0000    00 00 00 20 0e 00 80 ff  00 00 00 00               ... ........

RX: CplD:   Len: 020 ReqID: 0e00 CplID: 0000 Status: 0 BC: 080 Tag: 80 LowAddr: 00
0000    4a 00 00 20 00 00 00 80  0e 00 80 00 f3 ee 00 f0   J.. ............
0010    f3 ee 00 f0 c3 e2 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0020    54 ff 00 f0 4d 30 00 f0  f5 2f 00 f0 a5 fe 00 f0   T...M0.../......
0030    87 e9 00 f0 f3 ee 00 f0  f3 ee 00 f0 f3 ee 00 f0   ................
0040    f3 ee 00 f0 57 ef 00 f0  53 ff 00 f0 60 0b 00 c0   ....W...S...`...
0050    4d f8 00 f0 41 f8 00 f0  30 71 00 f0 39 e7 00 f0   M...A...0q..9...
0060    34 01 00 e8 2e e8 00 f0  d2 ef 00 f0 00 e0 00 f0   4...............
0070    f2 e6 00 f0 6e fe 00 f0  53 ff 00 f0 53 ff 00 f0   ....n...S...S...
0080    a4 f0 00 f0 c7 ef 00 f0  20 35 00 c0               ........ 5..

real    0m0.383s
user    0m0.004s
sys     0m0.015s

ufrisk, 2019-11-26:
Thanks for the update, then it's the auto-detect that takes an aweful amount of time on Linux, thats why I never caught it I guess - I mostly test on WIndows. I wonder if this extra latency/slowness issue is responsible for the somewhat lower performance on Linux as well. Anyway I'm happy the issue is resolved and I'm closing this issue.

closed 2019-11-26 by ufrisk