ufrisk/pcileech-fpga #40: Issuing the dump command twice with the same -out file causes segfault

asbridge, 2019-11-25:
Ulf,

If the dump command fails to open its -out file, looks like it frees the file anyway leading to a double free and this stack trace:

Vince

Do this command twice.  Second time it will segfault.
 ./pcileech -vvv dump -min 0x0 -max 0x8000 -force -out mem.dmp

Memory Dump: Failed. File already exists.
*** Error in `./pcileech': double free or corruption (!prev): 0x0000000000e2f1e0 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x81679)[0x7fa696a40679]
/lib64/libc.so.6(fclose+0x177)[0x7fa696a2d167]
./pcileech[0x40a91c]
./pcileech[0x40aec6]
./pcileech[0x40af1f]
./pcileech[0x403a40]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7fa6969e1505]
./pcileech[0x401ab9]
======= Memory map: ========
00400000-00424000 r-xp 00000000 00:29 99023389                           /vba/LambdaConcept/PCILeech/pcileech/files/pcileech
00623000-00624000 r--p 00023000 00:29 99023389                           /vba/LambdaConcept/PCILeech/pcileech/files/pcileech
00624000-00625000 rw-p 00024000 00:29 99023389                           /vba/LambdaConcept/PCILeech/pcileech/files/pcileech
00e2c000-02e99000 rw-p 00000000 00:00 0                                  [heap]
7fa690000000-7fa690021000 rw-p 00000000 00:00 0
7fa690021000-7fa694000000 ---p 00000000 00:00 0
7fa6947c5000-7fa6947c6000 ---p 00000000 00:00 0
7fa6947c6000-7fa694fc6000 rw-p 00000000 00:00 0
...

ufrisk, 2019-11-25:
Thanks for reporting this issue. It should now be updated in the pcileech project.

closed 2019-11-25 by ufrisk

asbridge, 2019-11-26:
Ulf,

Confirmed.  Fixed.  Thanks.

Vince