ufrisk/pcileech-fpga #43: Does the pci extension cable matter?

EelhandsMcCoy, 2019-12-19:
I bought the AC701 which is a pcie 2 / x4 connection, I believe. Can I make a 'bad' decision buying something off Amazon?

Would this: https://www.amazon.com/Timack-Express3-0-Extension-Gold-Plated-Connector/dp/B07QPF748K/ref=sr_1_6?keywords=PCIe%2B2.0%2Bx4%2Bextension%2Bcable&qid=1576715320&sr=8-6&th=1

Be fine? Or:

https://www.amazon.com/Express-Extender-Extension-Gold-Plated-Connector/dp/B07J2BJPDN/ref=sr_1_5?keywords=PCIe+extension+cable&qid=1576715670&sr=8-5

I'm sorry to ask such a trivial question.

ufrisk, 2019-12-19:
I dunno about the 1st cable, I don't recognize the connector. The 2nd cable may work, but it carries power as well - which is not ideal for the AC701 - I never had a problem myself with it, but others have had troubles with it.

For extension cables I recommend "bitcoin cables" such as this one: https://www.amazon.com/AAAwave-Version-Powered-Adapter-Extension/dp/B074GGNMSC

No need to power the adapter itself.

EelhandsMcCoy, 2019-12-20:
Thank you so much!

closed 2019-12-20 by EelhandsMcCoy