ufrisk/pcileech-fpga #45: Screamer M.2 - Unable to retrieve required Device PCIe ID

fa1Lure, 2020-01-09:
Hello, so I think I have successfully flashed my screamer, using linux. 
LD1 and LD3 are green.

When I run this on my windows machine I get the following errors;

`pcileech kmdload -kmd win10_x64 -device fpga -v`

```
FPGA: ERROR: Unable to retrieve required Device PCIe ID [0,v0.0,0000]
PCILEECH: Failed to connect to the device.
```

Best regards.

ufrisk, 2020-01-09:
when you say you can see it in device manager or not, do you mean the USB-side of things? i.e. you are replugging the USB cable? Or do you mean the PCIe side of things.

The `FPGA: ERROR: Unable to retrieve required Device PCIe ID [0,v0.0,0000]` my USB code somehow not being able to communicate with the device (but USB connection should be up).

The flashing does not look very nice, it have some bad errors. I have not tested flashing with OpenOCD. I have always used a Xilinx cable and the Xilinx flash tools for it. I'll try to check out OpenOCD this weekend to see whats possibly going on.

I'll also put it on my todo list to add some blinking patterns to the LEDs on startup so that we'll see if a bitstream is correctly flashed.

Please let me know if you're able to resolve, otherwise I'll be looking into it this weekend.

fa1Lure, 2020-01-09:
> when you say you can see it in device manager or not, do you mean the USB-side of things? i.e. you are replugging the USB cable? Or do you mean the PCIe side of things.
>
> The `FPGA: ERROR: Unable to [...]

I think the USB is fine, I've tried with different cables aswell. I can see them all in device manager.

The PCIe device shows up on my attacker pc but it does not show up in the device manager on my target PC.

I've tried flashing with both 4.1 and 4.2 on different OS and PC's with different builds of openocd. They all give the same output.

Can anyone who flashed with the JTAG sign in and tell me if it's working? JTAG and openocd.

ufrisk, 2020-01-09:
Thanks for the info, I'll try out flashing with OpenOCD this weekend and get back to you.

ufrisk, 2020-01-12:
Thanks for reporting this.

This indeed seems to be an issue. The instructions about how to flash at lambdaconcept will be updated next week. I'll post here when they're updated. There will be some changes in configuration and also you may need to use a linux-version of OpenOCD. But instructions will be posted next week.

fa1Lure, 2020-01-12:
> Thanks for reporting this.
> 
> This indeed seems to be an issue. The instructions about how to flash at lambdaconcept will be updated next week. I'll post here when they're updated. There will be some changes in configuration and also you may need to use a linux-version of OpenOCD. But instructions will be posted next week.
> 
> Meanwhile, if you do have access to a Xilinx programming cable it works fine.

Alright, thanks.

povauboin, 2020-01-14:
Hello, just to clarify:

If you have problems writing to the SPI flash using openOCD and LambdaConcept cable (typically "content differ" errors), you need to patch openOCD with the "flashid.patch" as explained in http://blog.lambdaconcept.com/doku.php?id=products:jtagserial_programmer#errorcontents_differ

If the write is successful but the FPGA does not boot (FPGA Prog Done LED LD3 is not green) you may have an issue with the Flash SPI Quad Enable bit. (This bit is required as pcileech image is built with SPI buswidth set to 4). This bit should have been set on the Screamer M.2 at the factory, but maybe your board got skipped in the process.

To solve this, you may either:
1. Build pcileech image without quad SPI buswidth by removing this line: https://github.com/ufrisk/pcileech-fpga/blob/ddad9b6ac9bafc5dfd7293b56519f7437ea1d33d/ScreamerM2/src/pcileech_screamer_m2.xdc#L194
2. Or use Xilinx cable and the Xilinx flash tools that will set the Quad bit in the flash for you.
3. Or patch openOCD as explained here: http://blog.lambdaconcept.com/doku.php?id=products:screamer_m2#errorthe_fpga_does_not_boot

I hope this helps

fa1Lure, 2020-01-15:
> Hello, just to clarify:
>
> If you have problems writing to the SPI flash using openOCD and LambdaConcept cable (typically "content differ" errors), you need to patch openOCD with the "flashid.patch" [...]

Hello, yes I did patch openOCD with flashid.patch because I got "content differ" errors before.
I have now tried patching openOCD as explained and I'm getting this output;

Open On-Chip Debugger 0.10.0+dev-01012-ged8fa09c (2020-01-15-00:34)
Licensed under GNU GPL v2
For bug reports, read
    http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : sector 0 took 80 ms
Info : sector 1 took 80 ms
Info : sector 2 took 69 ms
Info : sector 3 took 66 ms
Info : sector 4 took 70 ms
Info : sector 5 took 78 ms
Info : sector 6 took 75 ms
Info : sector 7 took 65 ms
Info : sector 8 took 77 ms
Info : sector 9 took 86 ms
Info : sector 10 took 85 ms
Info : sector 11 took 80 ms
Info : sector 12 took 83 ms
Info : sector 13 took 77 ms
Info : sector 14 took 74 ms
Info : sector 15 took 50 ms
Info : sector 16 took 63 ms
Info : sector 17 took 64 ms
Info : sector 18 took 62 ms
Info : sector 19 took 75 ms
Info : sector 20 took 68 ms
Info : sector 21 took 65 ms
Info : sector 22 took 68 ms
Info : sector 23 took 61 ms
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
av0 STATUS: 00000040
av STATUS: 00000042
ap STATUS: 00000040
shutdown command invoked

Wolrus, 2020-01-15:
Hi, I'm also having issues with the Screamer M.2.
I used openocd and it seems to be "working", the 3 LEDs are on but i'm not able to read any pages
```pcileech probe -v

 Memory Map:
 START              END               #PAGES

 Current Action: Probing Memory
 Access Mode:    Normal
 Progress:       4096 / 4096 (100%)
 Speed:          455 MB/s
 Address:        0x0000000100000000
 Pages read:     0 / 1048576 (0%)
 Pages failed:   1048576 (100%)
Memory Probe: Completed.```

However I'm able to see my custom PCIe configuration space using -vv,  the values seems to be correct, (device ID, vendor ID) so I think the flashing went fine.

Regarding the usb drivers, the instructions were not helpful, I'm supposed to use these ? (that's what I've been able to find googling "ftdi" drivers) https://www.ftdichip.com/Drivers/D3XX.htm

fa1Lure, 2020-01-17:
Ok, so I have tried flashing my screamer with the JTAG-HS2 cable and it says the flash was done successfully.
LED1 and LED3 are green but LED2 is still not green.
I've tried with the prebuilt binary and also building my own.

ufrisk, 2020-01-17:
> 
> 
> Ok, so I have tried flashing my screamer with the JTAG-HS2 cable and it says the flash was done successfully.
> LED1 and LED3 are green but LED2 is still not green.
> I've tried with the prebuilt binary and also building my own.

It's probably working as intended then. LED2 is only green when data is transmitted over USB as-is now. I'll hopefully some time next week publish a cosmetic update to the bitstream which will blink LED2 4 times at power-on just to give an easier way to tell if it's properly flashed.

fa1Lure, 2020-01-17:
> It's probably working as intended then. LED2 is only green when data is transmitted over USB as-is now. I'll hopefully some time next week publish a cosmetic update to the bitstream which will blink [...]

Alright. 
What could be the problem then? I'm still getting the same error.
I've tried with different pc's aswell. The device will only show in device manager after a cold boot.

`FPGA: ERROR: Unable to retrieve required Device PCIe ID [0,v0.0,0000]`
`PCILEECH: Failed to connect to the device.`

ufrisk, 2020-01-17:
> Alright.
> What could be the problem then? I'm still getting the same error.
> I've tried with different pc's aswell. The device will only show in device manager after a cold boot.
>
> `FPGA: ERROR: Unable [...]

It's most likely a separate issue from the flashing issue. Maybe it's similar to this? https://github.com/ufrisk/pcileech-fpga/issues/46  If you think it's a different issue please start a new issue about it since it's not really related to the flash error.

Anyway, I hope to publish the new bitstream next week which will make it super clear whether it's flashed or not at least.

povauboin, 2020-01-17:
As far as I can tell this is indeed not a flashing issue.

@fa1Lure @Wolrus : your openOCD output looks fine to me. ("Warn : device needs paging or 4-byte addresses - not implemented" are just warnings, not errors). And the FPGA boots correctly because you have LED3 green.

fa1Lure, 2020-01-18:
I've solved my problem, I reflashed my device and made sure that I had the latest FTD3XX drivers.

biGGer, 2020-01-24:
A bit of info in case someone else runs into the same problem as i did.
On cold boot LD1 and LD3 stay lit and i'm getting this(Yes i connected it to the same PC)
[image]
If i press the Prog button LD1 starts blinking and i started getting this
[image]
And then i did a reboot through Windows and it finally started working properly.

ufrisk, 2020-01-25:
I updated the PCILeech ScreamerM2 bitstream to version 4.3. It should now blink the LD2 led on the ScreamerM2 upon power-up. If the LD2 blinks after a flash that means the flashing itself was successful and should rule out any flashing issues.

Also, if it's possible to read internal device data that means USB communication (and flash) is working. Root cause is problem with the PCIe link in those cases.

rzfyxdbd, 2020-02-08:
Thanks, having LD2 blink on startup really helps!

I narrowed the problem to the ScreamerM2 not being properly flashed and ended up having to apply both patches to the latest openocd (flashid.patch & qe_command.patch) from the lambdaconcept website.

Only after doing so and flashing the device again I saw LD2 blinking.

ufrisk, 2020-07-08:
I'm closing this issue since it seems to be resolved.

Also, I suggest you upgrade to the newly released v4.4 of the bitstream for increased stability. If you do need instructions about how to flash LambdaConcept put up a nice new guide with pre-compiled OpenOCD versions even at: https://docs.lambdaconcept.com/screamer/

Good Luck :)

closed 2020-07-08 by ufrisk