ufrisk/pcileech-fpga #46: FPGA not detected/working with certian RAM Speeds

bin5k, 2020-01-12:
I recently decided to mess with X.M.P/Ram speeds and i noticed the FPGA stops working if i changed the ram speed to anything above 2700mhz.

Everything works fine with these ram speeds:
> 2133mhz (Default speed)
> 2666mhz
> 2700mhz
> 
> 2800mhz - The FPGA is not detected in Windows at all and reading memory doesn't work
> 3200mhz (X.M.P) - Nothing works.

Even with X.M.P Off, just manually choosing 2800mhz, nothing works at all.
I also tried different PCI-E slots and extension cables with the same result.

I'm wondering if I'm the only one having this issue or what ?

my setup is:

FPGA: AC701/FT601 (Bitstream 4.2, 4.0)
OS: Windows 10 / Build 1903
RAM: 32 GB DDR4 (Default Speed 2133mhz, X.M.P = 3200mhz)
CPU: Intel i9-9900K (default clocks)

gcakir, 2020-01-12:
- Did you try installing it into PCIe x1 slot?
- Did you try increasing (i.e. relaxing) the timing settings of RAM modules when you manually set 2800MHz?

Sometimes, it might require multiple restarts (full power cycles) until the OS sees it. In some cases, I'd first install it into x16 slot and start the computer but the OS wouldn't see it. Then I'd turn off the system completely and install it back into x1 slot, and then it would work.

There is this also: https://github.com/ufrisk/pcileech/issues/105

bin5k, 2020-01-12:
> 
> 
>     * Did you try installing it into PCIe x1 slot?
> 
>     * Did you try increasing (i.e. relaxing) the timing settings of RAM modules when you manually set 2800MHz?
> 
> 
> Sometimes, it might require multiple restarts (full power cycles) until the OS sees it. In some cases, I'd first install it into x16 slot and start the computer but the OS wouldn't see it. Then I'd turn off the system completely and install it back into x1 slot, and then it would work.
> 
> There is this also: [ufrisk/pcileech#105](https://github.com/ufrisk/pcileech/issues/105)

I tried multiple slots (x4, x16 and the x1) and the result is the same.
About the 2800mhz i also tried 2800mhz with auto timings and X.M.P timings and nothing is detected at all.

It's a weird issue because 2700mhz and bellow works just fine but 100mhz more and nothing works.

ufrisk, 2020-07-08:
can you please retry this with latest v4.4 fpga bitstream released just today to see if it resolves this issue?

closed 2020-09-17 by ufrisk