ufrisk/pcileech-fpga #47: Unable to retrieve PCIe ID only on latest versions.

sb0sse, 2020-02-16:
On version 4.4 and onwards (respective leechcore and vmm versions), even with latest bitstream, on SP605, I cannot use pcileech. I trace the issue to here:

```
if(!ctx->wDeviceId) {
        szDeviceError = "Unable to retrieve required Device PCIe ID";
        goto fail;
    }
```

I can't think of any legitimate reason. There is zero difference, version 4.3 (and leechcore/vmm libs from december 2019) works perfect but new versions from 2020 onwards are totally broken. It doesn't make a difference whether I'm interacting with pcileech itself of using the vmm/leechcore API.

ufrisk, 2020-02-16:
I checked this out and it works fine for me with the SP605 on the 4.4 release.

which version do you have on your `FTD3XX.dll` file?

[image]

Other than this there might be a physical connection error, try replug the usb cable and/or the ftdi extension board.

sb0sse, 2020-02-16:
@ufrisk 

On 4.3 (using FTDI v1.2.0.5):

```
DEVICE: FPGA: SP605 / FT601 PCIe gen1 x1 [400,175,500] [v2.2,0400]
Memory Display: Contents for address: 0x0000000000001000
0000    e9 4d 06 00 01 00 00 00  00 00 00 00 3f 00 18 10   .M..........?...
0010    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0020    00 00 00 00 00 00 00 00  00 00 00 00 00 9b 20 00   .............. .
0030    00 00 00 00 00 00 00 00  ff ff 00 00 00 93 cf 00   ................
0040    00 00 00 00 00 00 00 00  ff ff 00 00 00 9b cf 00   ................
0050    00 00 00 00 00 00 00 00  00 50 ff dc 00 00 00 00   .........P......
0060    7c 16 00 00 30 00 da 16  00 00 10 00 00 00 00 00   |...0...........
0070    e0 f4 6c 7c 02 f8 ff ff  00 70 00 c0 9f f7 ff ff   ..l|.....p......
0080    06 01 07 00 06 01 07 00  01 49 00 00 00 00 00 00   .........I......
0090    31 00 05 80 00 00 00 00  00 00 00 00 00 00 00 00   1...............
00a0    00 d0 1a 00 00 00 00 00  b8 0e 35 00 00 00 00 00   ..........5.....
00b0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00c0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00d0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00e0    00 00 00 00 00 00 57 00  b0 7f 49 f2 81 a3 ff ff   ......W.. I.....
00f0    00 00 00 00 00 00 ff 0f  00 50 49 f2 81 a3 ff ff   .........PI.....
```

On 4.4 (using FTDI v1.3.0.2):
```
DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [0,v0.0,0000]
PCILEECH: Failed to connect to the device.
```

Problem occurs on all PCIe slots, on all USB ports, rebooted, reseated, and swapped ports a million times.

ufrisk, 2020-02-17:
what happens if you use FTDI v1.3.0.2 on 4.3?

sb0sse, 2020-02-17:
4.3 with the latest FTDI driver (v1.3.0.2):
```
DEVICE: FPGA: SP605 / FT601 PCIe gen1 x1 [400,175,500] [v2.2,0a00]
Memory Display: Contents for address: 0x0000000000001000
0000    e9 4d 06 00 01 00 00 00  00 00 00 00 3f 00 18 10   .M..........?...
0010    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0020    00 00 00 00 00 00 00 00  00 00 00 00 00 9b 20 00   .............. .
0030    00 00 00 00 00 00 00 00  ff ff 00 00 00 93 cf 00   ................
0040    00 00 00 00 00 00 00 00  ff ff 00 00 00 9b cf 00   ................
0050    00 00 00 00 00 00 00 00  00 50 ff dc 00 00 00 00   .........P......
0060    7c 16 00 00 30 00 da 16  00 00 10 00 00 00 00 00   |...0...........
0070    e0 64 b7 6f 02 f8 ff ff  00 70 00 c0 9f f7 ff ff   .d.o.....p......
0080    06 01 07 00 06 01 07 00  01 49 00 00 00 00 00 00   .........I......
0090    31 00 05 80 00 00 00 00  00 00 00 00 00 00 00 00   1...............
00a0    00 d0 1a 00 00 00 00 00  b8 0e 35 00 00 00 00 00   ..........5.....
00b0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00c0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00d0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00e0    00 00 00 00 00 00 57 00  b0 7f 69 a9 81 d2 ff ff   ......W.. i.....
00f0    00 00 00 00 00 00 ff 0f  00 50 69 a9 81 d2 ff ff   .........Pi.....
```

Please note the PCIe ID change is due to a reflash to a different mcs, I did this to eliminate any chance of flashing problems. I have now used 3 custom mcs files (modified IDs; custom configurations) as well as the stock one to no effect.

ufrisk, 2020-02-17:
Thanks,

That PCIe ID is related to which PCIe slot it's connected to, it shouldn't be affected by flashing it.

I'll re-check this when I'm back (sadly some time next week most probably), but it's a bit strange since it where working for me when I checked it out yesterday, but I'll re-check again.

sb0sse, 2020-02-17:
@ufrisk Admittedly I am using the latest release, do you want me to try building the master branch?

sb0sse, 2020-05-15:
Hi Ulf,

This seems to still be a problem. I have now tested on multiple devices (SP605 + M.2 Screamer) across multiple systems (Intel X99, AMD Zen+, AMD Zen2). I noticed in another issue the screenshot of FTD3XX.dll you posted has a recent modified timestamp (2020) but the one I'm downloading (1.3.0.2) is from 2018, could this be the cause? Can you explicitly link to the FTD3XX.dll that you are using?

ufrisk, 2020-05-15:
I download from:
* https://github.com/ufrisk/pcileech/releases/download/v4.4/PCILeech_files_and_binaries_v4.4-20200316.zip
* https://www.ftdichip.com/Drivers/D3XX/FTD3XXLibrary_v1.3.0.4.zip

place 64-bit version of FTD3XX.dll v1.3.0.4 alongside pcileech.exe and go.

also there is a kernel driver from ftdichip for this; but it's automatically downloaded from windows update on first connect of a ftdi device. maybe there will be a problem if your pc isn't internet connected or if you have a very old driver (then force update in device manager or download driver and install manually from ftdichip.com).

ufrisk, 2020-07-08:
I suspect this issue is now resolved since I haven't heard back so I'll close this issue.

Also, I would suggest upgrading to v4.4 of the fpga bitstream relased just today for increased stability.

closed 2020-07-08 by ufrisk