ufrisk/pcileech-fpga #49: FPGA: ERROR: Unable to load FTD3XX.DLL

helloboiss, 2020-03-06:
I'm using PCIeScreamer v1 and haven't used for like half a year. Everything was fine back then but now I get this strange error: FPGA: ERROR: Unable to load FTD3XX.DLL [0,v0.0,0000]. Windows also don't detect the PCIeScreamer device with 0x666. 

Also flashed the device with old pre-binaries v3.2 and the new ones. Nothing works. FTD3XX.DLL is in the same map as pcileech.exe. LED is flashing while holding down the button on the PCIeScreamer, so I don't think that flashing is the problem.

I still have the same PC as when it worked so it is really strange.. you think my device is broken?

ufrisk, 2020-03-06:
have you downloaded the most recent FTD3XX.DLL from ftdichip.com and placed it alongside the other binaries? The version should be 1.3.0.2

[image]

helloboiss, 2020-03-06:
There was indeed some problem with my FTDI driver it was a WinUSB instead of FTDIBUS (I think i messsed something up with Zadig) . Now i get the following error:
FPGA: ERROR: Unable to retrieve required Device PCIe ID[1,v3.2,0000].

For me the v3.2 bitstream was stable like half year ago.. Also Windows still can't detect my fpga device.

ufrisk, 2020-03-06:
Now the problem indicates that there is some problem with the PCIe connection to your computer. The USB connection is fine since it detects the bitstream version. The PCIe bus:num is not detected though - which is a clear indication of some PCIe related error.

The PCIeScreamer R01 have known instabilities with regards to this.

If it was stable half a year ago can you please try to add it to the same PCIe slot as last time; also you can test upgrading it to the most recent bitstream I have. It may or may not help you. Unfortunately since the instabilities are related to the hardware itself I'm not really able to fix this in software.

helloboiss, 2020-03-06:
I have tested with the most recent bitstream. Also all bitstream with version 4.x gives me a flashing LED light. I downloaded the pre-builts with version R01.

ufrisk, 2020-03-06:
I'll double check this next week to see if there is an error on my side. Unfortunately chances are that this is related to known hardware issues with the PCIeScreamerR01 which I cannot do anything about.

helloboiss, 2020-03-12:
Tried on a different pc with another motherboard. Still the same results

ufrisk, 2020-08-09:
The unable to load FTD3XX.DLL should have been fixed in the last update. Closing issue.

closed 2020-08-09 by ufrisk