ufrisk/pcileech-fpga #50: NeTV2 Not connecting.

njbrown09, 2020-04-09:
I try to connect with pcileech but im getting "Unable to connect to USB/FT601 device"

ufrisk, 2020-04-09:
yes, but just posting "it's not working" without any information is worthless info for me to be able to give any help at all.

1. have you flashed the NeTV2?
2. can you ping the NeTV2?
3. what is your PCILeech command line?

njbrown09, 2020-04-09:
1. Yes, i flashed it using the build instructions.
2. Yes, it gets an ip address with DHCP and i can ping it in windows.
3. https://gyazo.com/392fb300693cf5e1093ae83568f133f4

njbrown09, 2020-04-09:
The NeTV2 is flashing the green light, and does not stop after 10 seconds (It does not go red). Id assume since it did get a dhcp address it is normal for the green flashing to never stop?

ufrisk, 2020-04-09:
`pcileech.exe probe -device rawudp://ip=192.168.0.156 -v`

njbrown09, 2020-04-09:
Thank you, i guess im dumb.

closed 2020-04-09 by njbrown09