ufrisk/pcileech-fpga #51: NeTV2: ERROR: Unable to retrieve required Device PCIe ID

nazarihome, 2020-04-18:
I am trying to use PCILeech with NeTV2. I flashed the fpga and inserted it into PCIe port. I am using the an Ethernet cable to connect the fpga board to my computer. The PCI status light is green and the ETH status is blinking red. I can ping the board: 
```
\PCILeech_files_and_binaries_v4.4-20200316>ping 192.168.0.222

Pinging 192.168.0.222 with 32 bytes of data:
Reply from 192.168.0.222: bytes=32 time<1ms TTL=128
Reply from 192.168.0.222: bytes=32 time<1ms TTL=128
Reply from 192.168.0.222: bytes=32 time<1ms TTL=128
```
However, when I try to dump by pcileech I get this error. 
```
D:\netv2\PCILeech_files_and_binaries_v4.4-20200316>pcileech.exe dump -device rawudp://ip=192.168.0.222 -v

DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [5,v4.2,0000]
PCILEECH: Failed to connect to the device.

D:\argus\netv2\PCILeech_files_and_binaries_v4.4-20200316>pcileech.exe dump -device rawudp://ip=192.168.0.222 -vv

----- FPGA DEVICE CONFIG REGISTERS: CORE-READ-ONLY     SIZE: 34 BYTES -----
0000    89 ab 00 00 22 00 00 00  04 02 05 00 00 00 00 00   ...."...........
0010    1f 7c fc 91 39 00 00 00  33 7c fc 91 39 00 00 00   .|..9...3|..9...
0020    00 00                                              ..

----- FPGA DEVICE CONFIG REGISTERS: CORE-READ-WRITE    SIZE: 30 BYTES -----
0000    cd ef 04 00 1e 00 00 00  a0 86 01 00 00 00 00 00   ................
0010    ee 10 07 00 ee 10 66 06  02 3c 00 00 00 00         ......f..<....

----- FPGA DEVICE CONFIG REGISTERS: PCIE-READ-ONLY     SIZE: 48 BYTES -----
0000    01 23 00 00 30 00 00 00  00 00 19 08 24 00 00 00   .#..0.......$...
0010    00 00 11 10 00 00 00 02  10 29 00 00 00 00 00 00   .........)......
0020    11 10 00 00 1d 7f 00 00  00 00 00 00 00 00 00 00   ..... ..........

----- FPGA DEVICE CONFIG REGISTERS: PCIE-READ-WRITE    SIZE: 84 BYTES -----
0000    45 67 00 f0 54 00 00 00  35 0a 00 01 01 00 00 00   Eg..T...5.......
0010    00 00 00 00 00 f0 48 00  00 00 00 0e 00 00 00 00   ......H.........
0020    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0030    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0040    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0050    00 00 00 00                                        ....

----- PCIe CORE Dynamic Reconfiguration Port (DRP)  SIZE: 0x100 BYTES -----
0000    00 00 00 01 00 02 00 00  00 00 00 00 00 00 f0 00   ................
0010    ff ff 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0020    00 00 00 00 00 00 00 00  00 00 00 40 00 00 00 00   ...........@....
0030    00 00 09 02 30 00 21 7f  02 00 01 00 00 03 11 0c   ....0.! ........
0040    02 a1 00 43 01 00 1f fd  7f ff 09 ff 01 20 01 48   ...C.... .... .H
0050    00 05 01 60 11 9c 00 00  00 00 00 00 00 00 00 00   ...`............
0060    00 00 10 60 00 02 40 21  00 40 3d 48 00 23 00 00   ...`..@!.@=H.#..
0070    00 00 00 00 00 00 00 00  00 00 00 00 00 15 00 01   ................
0080    00 01 00 00 00 01 00 00  00 01 00 00 00 01 00 00   ................
0090    00 01 00 00 00 01 00 00  00 00 00 00 00 00 00 00   ................
00a0    00 00 00 00 00 00 00 00  00 02 00 00 12 34 10 18   .............4..
00b0    00 0b 00 01 00 11 00 00  00 00 00 00 00 01 00 00   ................
00c0    00 28 00 41 ff ff ff ff  00 e0 00 00 80 08 00 22   .(.A..........."
00d0    03 ff 01 72 02 48 00 08  00 20 0e 04 fa ac 00 00   ...r.H... ......
00e0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00f0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................

----- PCIe CONFIGURATION SPACE (no user set values) SIZE: 0x200 BYTES -----
0000    ee 10 66 06 00 00 10 00  02 00 00 02 00 00 00 00   ..f.............
0010    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0020    00 00 00 00 00 00 00 00  00 00 00 00 ee 10 07 00   ................
0030    00 00 00 00 40 00 00 00  00 00 00 00 ff 01 00 00   ....@...........
0040    01 48 03 78 08 00 00 00  05 60 80 00 00 00 00 00   .H.x.....`......
0050    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0060    10 00 02 00 e1 8f 00 00  10 29 00 00 12 f4 03 00   .........)......
0070    00 00 11 10 00 00 00 00  00 00 00 00 00 00 00 00   ................
0080    00 00 00 00 02 00 00 00  00 00 00 00 00 00 00 00   ................
0090    02 00 01 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00a0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00b0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00c0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00d0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00e0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00f0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0100    03 00 c1 10 35 0a 00 01  01 00 00 00 00 00 00 00   ....5...........
0110    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0120    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0130    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0140    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0150    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0160    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0170    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0180    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0190    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01a0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01b0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01c0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01d0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01e0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01f0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................

PCILEECH: Failed to connect to the device.

```
I tried restarting the computer few times, but it didn't help. Am I missing a step to detect the board by the target computer? any suggestion how I can debug this?

ufrisk, 2020-04-19:
The NeTV2 is correctly flashed and PCILeech is able to communicate with it properly.

The `0000` in `[5,v4.2,0000]` indicates that there is a problem with the PCIe connection. I know others have had problems with this on other systems as well; I haven't been able to pinpoint why yet since I don't really have a good test system myself to test against and I haven't been too keen on purchasing a new computer for a lot of money to test on.

Most systems so far that reported issues are newer AMD gaming PC's; some people have reported success by changing the PCIe slot; other people have reported success by underclocking the memory (but I fail to see why this should make a difference, but it's been reported).

What kind of system do you have?

nazarihome, 2020-04-22:
My motherboard is msi Z170A GAMING and my processor is Intel core i-7, overclocking is off. I switched to another target computer(A Dell X-64 server) and it doesn't give me the error anymore.

ufrisk, 2020-07-08:
can you please retry-this with the v4.4 bitstream released just today? does it resolve this issue?

closed 2020-09-07 by ufrisk

ConnorMAD, 2023-07-08:
FPGA: ERROR: Unable to retrieve required Device PCIe ID [0,v0.0,0000]
can you help me @ufrisk

ufrisk, 2023-07-08:
@ConnorMAD there is an error with the connection to the FPGA. Either you have no network connection to it (firewall, wrong ip address?) or there is an issue with the flash.

Can you ping the NeTV2 device? How does your pcileech command line look like?

ConnorMAD, 2023-07-10:
@ufrisk I think I'm trying to compile a non-updated version of pcileech.
I'm studying the project https://github.com/EngineOwningSoftware/pcileech-webradar, but I've tried updating it to the current pcileech files, but it doesn't recognize my FPGA.

ConnorMAD, 2023-07-19:
is there any wiki showing how i can deliver pcileech in my program? to open it on the second pc? @ufrisk

ConnorMAD, 2023-07-19:
I made a mini 2d radar project with websocket, but I don't know how to start integrating with pcilech

ufrisk, 2023-07-21:
@ConnorMAD If you wish to program against PCILeech there are documentation either in the form of header files (C/C++), docs.rs (Rust), MemProcFS project Wiki (Python), JavaDoc (Java) or Examples (C/C++, C#, Python and Rust).

Also if you're more interested in games (which I'm not really) there are other projects floating around out there that use my tools.

I'm not up for providing extensive help, sorry :\ Also I'm staying away from games and focusing more in the forensic aspects of my project.

ConnorMAD, 2023-07-22:
I have no intention of continuing with games, I'm just learning.
I thought games would be a good start.

ufrisk, 2023-07-24:
No worries, yeah, it's probably a good place to start. It's just that I'm not really able to help with that. But there should be code floating around doing that. Also check out the memprocfs examples: https://github.com/ufrisk/MemProcFS/tree/master/vmm_example

Best wishes with everything!

ConnorMAD, 2023-08-12:
@ufrisk Sorry for the ignorance, But would I read the memory without using a DMA?

ufrisk, 2023-08-12:
@ConnorMAD alternative ways are to use a driver to dump the memory. Examples are DumpIt or WinPMEM.

Also, if you're running inside a VM such as Hyper-V, VMWare or QEMU you may also access memory from the host os.

ConnorMAD, 2023-08-12:
@ufrisk is there any method of using this over long distance?

ufrisk, 2023-08-12:
There is a LeechAgent and "MemProcFS remoting" which you could use. It's not built for this use case but it should work if you add the FTD3XX.dll to the folder of the LeechAgent. You would then connect over the network to the Agent which could then in turn connect to the memory via whatever method you wish, (driver or pcie dma hardware).

It's primarily focused on AD environments, but it should be possible to run it in non-ad environments as well. There is also an "insecure" more with no authentication at all (dangerous if not properly firewalled).

https://github.com/ufrisk/MemProcFS/wiki/_Remoting

ConnorMAD, 2023-08-12:
> There is a LeechAgent and "MemProcFS remoting" which you could use. It's not built for this use case but it should work if you add the FTD3XX.dll to the folder of the LeechAgent. You would then [...]

Another thing, @ufrisk I was able to connect to vmware and get the memory of the vm, but how could I do the opposite? Using the vm to get memory from the main computer?

ufrisk, 2023-08-12:
That is not possible unless you use something like the agent I was mentioning over the network between host/vm.

ConnorMAD, 2023-08-13:
I read the guide about leechcore and leechagent, I thought it was great, but I have a question, is there a way to do this locally? using two local computers?

ConnorMAD, 2023-08-13:
-remote localhost?
or use ipv4 address for connection?

ufrisk, 2023-08-13:
You'd need to specify the IP-address / host-name of the other computer. If you specify localhost it will be trying to connect to the same computer not the 2nd one.

ConnorMAD, 2023-08-13:
@ufrisk Sorry for the ignorance I would have to put for example setspn -S ipv4adress mycomputername?

ufrisk, 2023-08-13:
easiest would probably be if you start the agent in insecure interactive mode. from a command prompt: leechagent.exe -interactive -insecure

and then connect to it with pcileech.exe display -min 0x1000 -device fpga -remote rpc://insecure:<your_remote_ip>

ensure the firewall on the system with the agent is off or allows incoming connections to tcp/28473 (if I remember correctly).

ConnorMAD, 2023-08-13:
thanks, connection, but this show me:
DEVICE: ERROR: unable to locate the winpmem driver file ''.

ConnorMAD, 2023-08-13:
Thank you, I will be a sponsor in your project. I'm learning many things

ufrisk, 2023-08-14:
You'd have to run the agent in elevated adminidtrator mode.

You'd have to run it from C:

The winpmem_x64.sys must be in the same folder as leechagent.exe

Please let me know how it goes :)

ConnorMAD, 2023-08-14:
Failed: RPC: RpcServerUseProtseqEpA (0x000006cc).

what is this error?
it was working normally, but I had to change computers and it didn't work again.
port 28473 is open TCP

ConnorMAD, 2023-08-14:
@ufrisk Can you help me with the above error?