ufrisk/pcileech-fpga #54: Device needs paging or 4-byte addresses

zredaxz, 2020-05-13:
Hi,

I just flashed my Screamer M.2 device with the latest openocd.
During the flashing I got this warning message: "WARN: Device needs paging or 4-byte addresses - not implemented".

Is this warning safe to ignore? Sorry if this is FAQ.
[image]

Thank you for your help!

ufrisk, 2020-05-13:
I don't use OpenOCD myself so I don't really know. I know there is a special version of the OpenOCD to be able to flash the ScreamerM2 though due to an otherwise unsupported memory.

Try flash it; if the device starts afterwards (blinks on LD2) and the output from pcileech when running `pcileech display -min 0x1000 -v -vv -device fpga` shows the new bitstream version it's working. Otherwise not.

zredaxz, 2020-05-13:
OK, I see, I will try this one more time. :)
If not OpenOCD, what is your recommendation?

Thank you for your help and this amazing tool!

ufrisk, 2020-05-13:
OpenOCD should be fine really, it's just that I use another more expensive programming cable which is supported by Xilinx Vivado directly when doing development.

But try it again, ignore the errors and see if the end result is good; otherwise re-check the instructions.

zredaxz, 2020-05-13:
Here is the output of the `pcileech display -min 0x1000 -v -vv -device fpga`

```
DEVICE: FPGA: ScreamerM2 PCIe gen2 x1 [300,0,500] [v4.3,3a00]

----- FPGA DEVICE CONFIG REGISTERS: CORE-READ-ONLY     SIZE: 34 BYTES -----
0000    89 ab 00 00 22 00 00 00  04 03 04 00 00 00 00 00   ...."...........
0010    28 5e d2 68 06 00 00 00  2a 5e d2 68 06 00 00 00   (^.h....*^.h....
0020    00 00                                              ..

----- FPGA DEVICE CONFIG REGISTERS: CORE-READ-WRITE    SIZE: 30 BYTES -----
0000    cd ef 04 00 1e 00 00 00  a0 86 01 00 00 00 00 00   ................
0010    ee 10 07 00 ee 10 66 06  02 3c 00 00 7f 00         ......f..<.. .

----- FPGA DEVICE CONFIG REGISTERS: PCIE-READ-ONLY     SIZE: 48 BYTES -----
0000    01 23 00 00 30 00 00 00  3a 00 16 08 7c 00 00 00   .#..0...:...|...
0010    00 00 00 00 00 00 00 00  30 29 00 00 00 00 40 00   ........0)....@.
0020    12 10 00 00 1e 7f 00 00  00 00 ff f9 00 00 00 00   ..... ..........

----- FPGA DEVICE CONFIG REGISTERS: PCIE-READ-WRITE    SIZE: 84 BYTES -----
0000    45 67 00 f0 54 00 00 00  35 0a 00 01 01 00 00 00   Eg..T...5.......
0010    00 00 00 00 ff fd 48 00  00 00 00 0e 00 00 00 00   ......H.........
0020    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0030    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0040    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0050    00 00 00 00                                        ....

----- PCIe CORE Dynamic Reconfiguration Port (DRP)  SIZE: 0x100 BYTES -----
0000    00 00 00 01 00 02 00 00  00 00 00 00 00 00 f0 00   ................
0010    ff ff 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0020    00 00 00 00 00 00 00 00  00 00 00 40 00 00 00 00   ...........@....
0030    00 00 09 02 30 00 22 7f  02 00 01 00 00 03 11 0c   ....0." ........
0040    02 a1 00 43 01 00 1f fd  7f ff 09 ff 01 20 01 48   ...C.... .... .H
0050    00 05 01 60 11 9c 00 00  00 00 00 00 00 00 00 00   ...`............
0060    00 00 10 60 00 02 40 21  00 40 3d 48 00 23 00 00   ...`..@!.@=H.#..
0070    00 00 00 00 00 00 00 00  00 00 00 00 00 15 00 01   ................
0080    00 01 00 00 00 01 00 00  00 01 00 00 00 01 00 00   ................
0090    00 01 00 00 00 01 00 00  00 00 00 00 00 00 00 00   ................
00a0    00 00 00 00 00 00 00 00  00 02 00 00 12 34 10 18   .............4..
00b0    00 0b 00 01 00 11 00 00  00 00 00 00 00 01 00 00   ................
00c0    00 28 00 41 ff ff ff ff  00 e0 00 00 80 08 00 22   .(.A..........."
00d0    07 ff 03 52 02 48 00 08  00 40 0e 84 fa ac 00 00   ...R.H...@......
00e0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00f0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................

----- PCIe CONFIGURATION SPACE (no user set values) SIZE: 0x200 BYTES -----
0000    00 00 00 00 00 00 10 00  02 00 00 02 00 00 00 00   ................
0010    00 00 10 c8 00 00 00 00  00 00 00 00 00 00 00 00   ................
0020    00 00 00 00 00 00 00 00  00 00 00 00 ee 10 07 00   ................
0030    00 00 00 00 40 00 00 00  00 00 00 00 ff 01 00 00   ....@...........
0040    01 48 03 78 08 00 00 00  05 60 80 00 00 00 00 00   .H.x.....`......
0050    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0060    10 00 02 00 30 29 00 00  00 00 00 00 12 f4 03 00   ....0)..........
0070    40 00 12 10 00 00 00 00  00 00 00 00 00 00 00 00   @...............
0080    00 00 00 00 02 00 00 00  00 00 00 00 00 00 00 00   ................
0090    02 00 01 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00a0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00b0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00c0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00d0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00e0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00f0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0100    03 00 c1 10 35 0a 00 01  01 00 00 00 00 00 00 00   ....5...........
0110    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0120    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0130    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0140    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0150    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0160    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0170    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
[... 4 lines cut ...]
01c0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01d0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01e0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
01f0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................

Successfully loaded LeechCore v1.7.0 Device 3
Memory Display: Contents for address: 0x0000000000001000
0000    e9 4d 06 00 01 00 00 00  01 00 00 00 3f 00 18 10   .M..........?...
0010    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
0020    00 00 00 00 00 00 00 00  00 00 00 00 00 9b 20 00   .............. .
0030    00 00 00 00 00 00 00 00  ff ff 00 00 00 93 cf 00   ................
0040    00 00 00 00 00 00 00 00  ff ff 00 00 00 9b cf 00   ................
[... cut ...]
0090    33 00 05 80 00 00 00 00  00 00 00 00 00 00 00 00   3...............
00a0    00 d0 1a 00 00 00 00 00  f8 06 35 00 00 00 00 00   ..........5.....
00b0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00c0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00d0    00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
00e0    00 00 00 00 00 00 57 00  b0 5f a1 8b 81 98 ff ff   ......W.._......
00f0    00 00 00 00 00 00 ff 0f  00 30 a1 8b 81 98 ff ff   .........0......

```

ufrisk, 2020-05-13:
The line: `DEVICE: FPGA: ScreamerM2 PCIe gen2 x1 [300,0,500] [v4.3,3a00]` tells you that you are running v4.3 of the bitstream. If that was your goal with flashing it succeeded.

zredaxz, 2020-05-14:
Almost. :)
From flashing perspective this was the plan. The original issue is it looks like 99% of the dump is failing.
The USB connection looks good, the PCI connections looks good. Fast boot is turned off, as well as IOMMU, KDMA, VBS, HVCI. So I am not sure what's going on.

ufrisk, 2020-05-15:
If it's working, but very badly, this may be due to bad signal quality over the PCIe link (if using long adapter cable for example) or it may be due to too aggressive built-in default timings in PCILeech.

https://github.com/ufrisk/LeechCore/wiki/Device_FPGA

You can try something like:

`pcileech.exe -v -vv dump -device fpga://pciegen=1,tmread=1500,tmwrite=1500,tmprobe=1500,readsize=0x10000,readretry=1`

to have a super conservative dump approach with 1 retry on each fail. It will be very slow if it works though; but adjust values to better fit your target system if this helps...

zredaxz, 2020-05-17:
> If it's working, but very badly, this may be due to bad signal quality over the PCIe link (if using long adapter cable for example) or it may be due to too aggressive built-in default timings in [...]

I replaced the cable with a shorter one and I tried to shield it. After the cable is replacement and with the proposed command above the overall result looks much better. The success rate is much higher but like you mentioned the speed is very slow it is ~25MB/s.

Looks like bunch of Bad PCIe TLP received error message is logged. What is this error message?

[image]

Based on your experience which device is more reliable, Screamer M.2 or the sp605? My sp605 is coming early next week.

Thank you!

ufrisk, 2020-05-17:
If you're able to put it directly into the PCIe slot thats the best, if using an adapter cable a very short cable is the best if using a non-shielded flat cable, if using a shielded cable it's generally working better. You can also try different slots. Also try another slot. Also PCILeech have known issues on some AMD Ryzen systems (which I haven't been able to pinpoint due to lack of access to such a test system).

Also, you may try an even more conservative setting such as:

`pcileech.exe -v -vv dump -device fpga://pciegen=1,tmread=2500,tmwrite=2500,tmprobe=2500,readsize=0x8000,readretry=1`

About adapters, if using them to connect to the wifi for example an [adapter like this is usually quite ok](https://www.ebay.co.uk/itm/ADT-Link-M-2-WiFi-A-E-key-Interface-Transfer-Extension-Cable-Support-PCI-E-3-0x1/274175302543), while an [adapter like this](https://www.ebay.co.uk/itm/NGFF-A-E-to-Mini-PCI-E-Adapter-M-2-WIFI-Adapter-Card-Board-Converter-for-C8B6/353053528492) is quite bad (use shortest possible cable).

zredaxz, 2020-05-17:
> If you're able to put it directly into the PCIe slot thats the best, if using an adapter cable a very short cable is the best if using a non-shielded flat cable, if using a shielded cable it's [...]

Super useful thank you! I ordered these now.
I can dump the memory with no problem. So looks like the PCIe connection reliable enough. IOMMU disabled.
However loading the win10_x64_2 is failing. I had the same issue few months ago and I gave up to figure this out:
`MemProcFS: Failed to initialize memory process file system in call to vmm.dll!VMMDLL_Initialize
KMD: Failed initializing required MemProcFS/vmm.dll
PCILEECH: Failed to load kernel module.`

Any idea why is it failing? After this error any subsequent request is failing with: `PCILEECH: Failed to connect to the device.`. The only resolution is rebooting the target machine. It is weird.

Regarding the AMD Ryzen issue: my Ryzen based target is not even booting with Screamer. :( Is this the error what you are troubleshooting? In this case I have repro machine...

zredaxz, 2020-05-22:
Closing this.
I tried with SP605 and everything is working as expected with no error. Looks like SP605 is significantly more reliable.

Thank you!

closed 2020-05-22 by zredaxz