ufrisk/pcileech-fpga #56: HS2 flash error- Program cannot be empty- program_hw_cfgmem

sledge-kege, 2020-05-29:
Hi Ulf,
I ran into this issue when flashing  the screamer m.2 with the HS2. I researched and tried a few things (path lengths and spaces in the paths) but I was wondering if anyone has come across this issue. Curranty on Windows 10 1903 and going to try Linux tomorrow.
 
Thanks for you time. 
Kedge

source vivado_flash_hs2.tcl -notrace
-------------------------------------------------------
 SEARCHING FOR PROGRAMMING CABLE AND OPENING DEVICE ...
-------------------------------------------------------
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository 'C:/Xilinx/Vivado/2019.2/data/ip'.
open_hw_manager: Time (s): cpu = 00:00:14 ; elapsed = 00:00:08 . Memory (MB): peak = 1010.125 ; gain = 337.828
INFO: [Labtools 27-2285] Connecting to hw_server url TCP:localhost:3121
INFO: [Labtools 27-2222] Launching hw_server...
INFO: [Labtools 27-2221] Launch Output:

****** Xilinx hw_server v2019.2
  **** Build date : Nov  6 2019 at 22:12:23
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

Warning: Cannot create '3000:arm' GDB server: An attempt was made to access a socket in a way forbidden by its access permissions

INFO: [Labtools 27-3415] Connecting to cs_server url TCP:localhost:3042
INFO: [Labtools 27-3417] Launching cs_server...
INFO: [Labtools 27-2221] Launch Output:

****** Xilinx cs_server v2019.2.0
  **** Build date : Nov 07 2019-00:28:22
    ** Copyright 2017-2019 Xilinx, Inc. All Rights Reserved.

connect_hw_server: Time (s): cpu = 00:00:02 ; elapsed = 00:00:08 . Memory (MB): peak = 1028.617 ; gain = 18.492
INFO: [Labtoolstcl 44-466] Opening hw_target localhost:3121/xilinx_tcf/Digilent/210249AE53B7
INFO: [Labtools 27-3164] End of startup status: HIGH
-------------------------------------------------------
 STARTING TO PROGRAM DEVICE ...                        
 THIS MAY TAKE SOME TIME.                              
-------------------------------------------------------
Program File cannot be empty
ERROR: [Labtools 27-3347] Flash Programming Unsuccessful: Program File cannot be empty
ERROR: [Common 17-39] 'program_hw_cfgmem' failed due to earlier errors.

    while executing
"program_hw_cfgmem -hw_cfgmem $my_hw_cfgmem"
    (file "vivado_flash_hs2.tcl" line 29)_

sledge-kege, 2020-05-29:
So this can be closed. 
I was able to flash with Linux, however I had to change line 9 in the "vivado_flash_hs2.tcl" from "open_hw" to "open_hw_manager"
The program was throwing a fit saying open_hw was depreciated. 
Once I changed that it seemed to work without an issue. 
Thanks for all the work
Kedge.

ufrisk, 2020-05-29:
Thanks for the update and I'm glad the issue is resolved.

As you already noticed it's been a while since I actually used that file myself; I mostly use the vivado GUI nowdays myself; I'll update the script to reflect the new Xilinx syntax for it. Thanks for reporting :)

ufrisk, 2020-07-08:
This should now be resolved - please try the new v4.4 bitstream to test. Thanks for reporting. I'll close this issue now.

closed 2020-07-08 by ufrisk