ufrisk/pcileech-fpga #57: Blinking PCI LED on PCIeScreamer R01/R02

synacktiv-qs, 2020-06-03:
Hello,

I am having trouble with PCIeScreamer R01 and R02, the PCI link LED (LD2) blinks and the device is not detected by my laptop and I get the error `DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [1,v4.2,0000]`.

I also tried with an external power supply without success.

I am using a PE3A to do the connection between the screamer and the Thunderbolt port. This setup is working fine with the SP605.

Flashing should be good, the test button lights up the LED and it is working fine on my desktop when I plug it directly into a PCI port.

I did not find any information about the blinking PCI link LED, do you have any idea what could be the cause?

Thanks for your time.

ufrisk, 2020-06-19:
Apologies for the slow answer. This one kind of got lost in the inbox. Have you been able to resolve the issue?

I'm aware that the artix-7 designs have some issues on some systems; like Ryzen 3900x and probably your Thunderbolt configuration as well. Problem is that I haven't been able to pinpoint this due to lack of testing equipment. 

I suspect it may be related to timing issues and PCIe core configuration. But due to the lack of a test system I haven't really been looking into this. Also, I haven't been too keep on buying a Ryzen 3900x that I only would need for testing...

involuntarysoul, 2020-07-11:
I have the same issue `DEVICE: FPGA: ERROR: Unable to retrieve required Device PCIe ID [4,v4.4,0000]
I am using ScreamerM2 r04 connected to thunderbolt 3. For me the PCI link led is solid 

[image]

ufrisk, 2020-07-11:
can you please try to update the firmware of your ScreamerM2 to the most recent version. this may or may not fix some of the stability issues you are experiencing.

involuntarysoul, 2020-07-11:
just flashed to 4.6 same issue persist, i disabled thunderbolt security from bios, is there some windows setting i should apply?

synacktiv-qs, 2020-07-16:
Hello,

Sorry for the delay in the reply. 
I had the same problem with the M2 Screamer (blinking PCI LED) and I was awaiting another adapter which does PCI -> Thunderbolt directly.

With this new adapter, the LED stays solid on the R02 and M2 and PCILeech works fine. The LED stays solid on the R01 but it does not work (cannot get the Device PCIe ID).

It seems that the problem was between the PE3A and the Screamers (even though the PE3A works fine with the SP605).

FYI: it was not on an AMD platform, it was against an old macbook air.

ufrisk, 2020-07-16:
The PCIeScreamer R01 and R02 does not route a specific signal I require for extra stability. It may be possible for me to do some additional workarounds; but it will never become as good as the SP605, AC701 or ScreamerM2 boards due to this hardware limitation. Also supporting multiple bitstreams and configs are extra work for me, and since this is a hobby project time is unfortunately limited. These are the reason why I dropped support in future bitstreams for the R01 and R02. 

It would be interesting to know if this problem persists for the ScreamerM2 if upgraded to the most recent bitstream versions. I know there are more issues at play here, but it should at least become similar to the SP605 with regards to stability.

synacktiv-qs, 2020-07-16:
I tested with bitstream v4.6 and also got this problem (LD1 blinking on the M2). I think I tried all available bitstreams for the M2 and had this problem.

Is there a way to debug this on my side?

ufrisk, 2020-07-16:
Blinking LED means that it have PCIe CLK but the that PCIe core is not fully initialized. This may be due to multiple factors including the host not yet initializing it.

You're a bit on your own when it comes to debugging; if using a M.2 slot you could try to increase # PCIe lanes from 1 to 4 (and also lowering max PCIe gen to 1 to keep same 64-bit output signal). This may be some work involved.

Also; in the most recent 4.6 version of the bitstream I added the possibility to trigger a global reset of the device by issing the following commands in device_fpga.c This would be the easier approach to try first.

```C
BYTE pbCoreResetSYS[] = { 0x00, 0x80 };
DeviceFPGA_ConfigWriteEx(ctx, 0x0002, pbCoreResetSYS, pbCoreResetSYS, FPGA_CONFIG_CORE | FPGA_CONFIG_SPACE_READWRITE);
```

Keep in mind that after such a reset you would need to re-initialize the FTDI code as well since the connection would be lost. Just reconnect after the global system reset. If lucky the core might come online properly after such a reset.

synacktiv-qs, 2020-07-28:
I had no luck with the global reset;
I will not be able to debug this further, I might get my hands on another PE3A to test, I will let you know if it works.

ufrisk, 2020-07-29:
Awesome, please let me know if it works. I have a few things I need to look into around this; possibly increasing the number of PCIe lanes I use in the design from 1 to the available 4 lanes. But it's quite some work involved so I don't expect to do this in the very near future.

Lets hope it's a faulty adapter.

closed 2020-09-03 by ufrisk