ufrisk/pcileech-fpga #58: Failed to launch run 'impl_1' due to failures in the following run(s): bram_pcie_cfgspace_synth_1

xNarkon, 2020-06-11:
Hello,

I have a problem while trying to build firmware ... I'm using Vivado 2020.1

I customized the Vendor and Device Identifiers by following: 
[image]

And while building after a while I'm getting this error:
`ERROR: [Common 17-70] Application Exception: Failed to launch run 'impl_1' due to failures in the following run(s):
bram_pcie_cfgspace_synth_1
These failed run(s) need to be reset prior to launching 'impl_1' again.`

What can be wrong?

ufrisk, 2020-06-11:
It may be due to the wizard unchecking some stuff on other pages - like the config space things when altering the values.

[image]

xNarkon, 2020-06-12:
Thanks for reply @ufrisk 

I think it's some kind different situation, I don't why but it cannot find this bram_pcie_cfgspace
[image]

The only thing I changed in Vivado 2020.1 Wizard were these ids :/

xNarkon, 2020-06-12:
I found an issue, running bitstream building from Vivado itself is somehow broken, when made it run from the console, everything went flawlessly.

closed 2020-06-12 by xNarkon