ufrisk/pcileech-fpga #59: ScreamerM2 - cannot flash with openocd

somethingisgo, 2020-06-12:
I purchased SCREAMER M.2 & JTAG SERIAL PACK from LambdaConcept.When using openocd to flash on win10,there is the following error:
Error: JTAG scan chain interrogation failed:all ones.
[image]

ufrisk, 2020-06-12:
You will need a special version of OpenOCD to flash the ScreamerM2 as described here and also in one of the other issue threads and on LambdaConcept page.

Also, I'm not really affiliated with LambdaConcept other than supporting the hardware the same way as I support the Xilinx dev boards, the USB3380 or the NeTV2 board so I'll have a really hard time helping you on this one :/

I use a HS2 JTAG cable compatible with Vivado myself so I'm not really that familiar with OpenOCD unfortunately.

Beercow, 2020-06-24:
Interesting. I ran into the exact same thing. Not sure how to troubleshoot it.

ufrisk, 2020-07-08:
LambdaConcept put up new instructions about how to flash with pre-compiled versions at https://docs.lambdaconcept.com/screamer/

Since this is not directly related to PCILeech itself I'm closing this issue. Good Luck with the flashing and I do recommend updating to v4.4 of the FPGA firmware if possible :)

closed 2020-07-08 by ufrisk

Beercow, 2020-07-08:
Thanks for the update. I think the my problem was I only had the programming cable plugged in. 😂