ufrisk/pcileech-fpga #60: DEVICE: FPGA: ERROR: Unable to connect to USB/FT601 device [0,v0.0,0000]

uranusb, 2020-06-20:
Hello,

I am using the NETV2 board and already flashed it successfully. I am able to ping the NeTV board. But when I run the following command:

E:\PCILeech>pcileech probe -device rawudp://192.168.0.222 -v

I got this error:

DEVICE: FPGA: ERROR: Unable to connect to USB/FT601 device [0,v0.0,0000]
PCILEECH: Failed to connect to the device.

I don't know why did we get this error because we are not using USB to connect the NeTV2 board. Please help us to solve this problem. Thanks

ufrisk, 2020-06-20:
please try:

`E:\PCILeech>pcileech probe -device rawudp://ip=192.168.0.222 -v`

uranusb, 2020-06-22:
Issue resolved, thanks!

ufrisk, 2020-06-23:
awesome, and good luck with your dma research :)

I'm closing this issue now since it's resolved.

closed 2020-06-23 by ufrisk