ufrisk/pcileech-fpga #61: NeTV2; NIC won't grap IP address

BUDDGAF, 2020-06-21:
The LED's that indicate weather it has a static or dhcp address seem to work fine. On boot the light is red if there is no cable, and green if there is as expected. Yet the thing just won't seem to take an IP address. The RasPi ethernet get's an address and I'm able to ssh into it as expected. One of the LED's on the NIC itself stays lit, wire or not, whenever the unit has power. Using a mirrored port on my switch, and then later directly connecting it to my desktops ethernet port using and using wireshark I'm able to see a little bit of udp traffic? possibly? but never anything reporting to be 192.168.0.222, or even anything in the 192.168 address space. 

I'm able to use the fltr program that came preloaded on the pi to get some data from the NeTV2 which hopefully means it's ok. That data looks like this.

pi@netv2mvp:~/code/flterm $ ./flterm --port /dev/ttyS0 --speed 115200 --kernel firmware.bin [FLTERM] Starting... adc93ddr:ad: 45Mwrit Mbpsl:Mbpsz) {"Rx_hr: 0, dmi_res 0, "_Rx_loc: 0,dmx_pha" : 0"dmi_sl_sy : , "h_Rx__pos"0", "i_Rxmbolors "00",rlayhres0, rlaes" :0, "eralyixeck" :, "orlay_ "0 0 0lay_bol_sync "ovy_spos"8", "rlamborors""00", "ddrps"5, "_write: 0, mye_opnin"0 00",rlay_eyeng" : 0 0", fpga_d_t "77C" json on Enabling json status RUNTIME> RUNTIME>input0: 0x0 @ 0. 0 MHz) input1: 0x0 (@ 0. 0 MHz) xadc: 77639 mC ddr: read: 145Mbs write: 0Mbps all: 45Mbps {hdmRx_hres" :0, "dmi_Rx_vres" : , "hdmi_Rx_pixel_clock" : , "hmiRx_pase" : 0 0 0", "hdmiRx_symbol_sync" 000 "hmi_Rx_snc_s" "0 0 0",hdmi_Rx_symbol_errrs" : "0 0 , "orlay_hres" : 0, "overlay_vres" : 0, "overaly_pixel_clock" : 0"ovelay_phase" "0 0","orlay_symol_snc : 111, "ovlay_syn_po" :"8 8 8" "overly_symbol_errrs" "00 0", "ddr_read_bps" : 15, "dr_rite_Mbs" : 0, "hdmix_ey_opeing" :"0 00" "overlay_ey_opening": "0 0 0","fpga_dietemp" : 77C" }

ufrisk, 2020-06-21:
The lights on the ethernet connection is not working in my PCILeech design - so what you are describing is normal.

If the red light for the network connection is lit that means that you have the default static IP address of 192.168.0.222.

If the green light for the network connection is lit that means that you got an IP address from your DHCP server. That's not necessarily in the 192.168-range - that is totally dependent on your DHCP config. Best is to check on the DHCP server which IP address got assigned.

Or if you use wireshark against the mirror port I guess you'll see the DHCP packets which assigns the IP address to the NeTV2...

BUDDGAF, 2020-06-21:
The lights and DHCP is confusing part. I'm aware of how it's supposed to work. What I'm saying is that the light turns green indicating that the NIC is being issued an IP address but _that isn't happening._ Let me get a wireshark capture to show you. I'll upload it here in a few minutes.

BUDDGAF, 2020-06-21:
[pcap.txt](https://github.com/ufrisk/pcileech-fpga/files/4809765/pcap.txt)

I enabled connection sharing on this pc then connected the NeTV2's ethernet port to a usb ethernet adapter. Then I started the packet capture, then rebooted the NeTV2. I let it run for a few minutes. I expect to see my PC issue the NeTV2 an ip address of 192.168.137.X. I don't see that I also see 0 transmissions from the NeTV2.

ufrisk, 2020-06-21:
I'm sorry, but the pcap doesn't say anything useful.

Can't you just run it in static mode; i.e. no DHCP so it gets the address 192.168.0.222 and then you put your host on the same local subnet?

Or check the DHCP server.

BUDDGAF, 2020-06-21:
No, I can't run it in static mode because it doesn't get an ip address. This is what I've been saying. The NIC doesn't work. I want to use it for DMA is it possible to use a usb network adapter, or get the data some other way like usb?

ufrisk, 2020-06-23:
yes, you should be able to use an usb network adapter just like any ordinary network adapter. if connecting directly you may need a "crossover network cable" depending on your network adapter though; most adapters auto-detect nowdays but I can't guarantee yours will do. But if it got an IP-address from your DHCP network connectivity should be working and you should be able to ping it if you do know which address it got.

ufrisk, 2020-08-09:
I'm closing this issue since I haven't heard anything in a while. I'm assuming it's fixed. It it's not please let me know and I'll re-open the issue.

closed 2020-08-09 by ufrisk