ufrisk/pcileech-fpga #62: Screamer M.2 Flashing Cable Connection Problem

Jackyl91, 2020-06-23:
Hi @Ufrisk,

I purchased both HS2 Diligent programming cable and LambdaConcept programming cable for flashing the Screamer M.2.  

I want to know the correct cable connection especially the HS2 cable as I failed flashing by both cables. 

Thank you.

ufrisk, 2020-06-23:
the pinout for the ScreamerM2 and the HS2 cable are both clearly marked on both ScreamerM2 and the HS2 respectively. They are compatible.

You will need to have to power the ScreamerM2 from the PCIe side when flashing it though. Just powering it over JTAG is not sufficient.

Jackyl91, 2020-06-30:
[image]

Thank you ufrisk. I finally flashed the ScreamerM2 by HS2 cable shown as the above-attached photo.  Is that necessary to generate project files after flashed the ScreamerM2? 
I have some trouble in using PCILeech with the result [PCILEECH: Failed to connect to the device.].

ufrisk, 2020-06-30:
You can flash with the pre-built bitstream which you may download in this project. Afterwards you can check if the flash is working if the LEDs are blinking on start-up as described in this project also.

You may use this with PCILeech, you can download the PCILeech binaries, but you will need to download a .dll file from ftdichip.com for it to work. To enable more debug output you can  try running; `pcileech.exe display -min 0x1000 -v -vv -device fpga`

ufrisk, 2020-07-08:
I'm closing this issue since you seems to have successfully re-flashed the device. I would suggest you upgrade to version 4.4 of the bitstream for better stability though.

On a successful flash LD2 will blink at power-on.

Good Luck :)

closed 2020-07-08 by ufrisk