ufrisk/pcileech-fpga #64: Intel i7 8750h FPGA ERROR

involuntarysoul, 2020-07-12:
The ScreamerM2 r04 running on 4.6 firmware was tested on desktop and functions 100%

However, installed on Razer Blade 15 with Intel i7 8750h, the screamer cannot be connected. Both PCIe and FPGA lights are solid green. When the test was running the USB light would flash. Something on the laptop is blocking the screamer. VT-d / secure boot / fastboot have all being disabled

[image]

[image]

involuntarysoul, 2020-07-12:
able to connect to screamer only under BIOS. once windows load it would fail, tested on both 1909 and 2004

[image]

ufrisk, 2020-07-12:
is your follow-up issue related to the Razer Blade as well or do it apply to your desktop computer?

involuntarysoul, 2020-07-12:
only Razer Blade, desktop has no issue

ufrisk, 2020-07-12:
I'll look into this; unfortunately there may be some time before I'll be able to do (like some months) this since I'm working on finishing the general update for release now. The "good" news is that I have a similar issue with my hades canyon so hopefully I'll be able to resolve your issue at the same time. I have some ideas what it may be about.

involuntarysoul, 2020-07-12:
After playing around with unlocked bios I am able to get screamer connected successfully after windows boot. It as something to do with disable native thunderbolt OS support. Originally it was set to win 10. Other than failed initialization the screamer can only read 4BG of RAM (out of 64gb) and cannot read any page files

[image]

ufrisk, 2020-07-16:
What you are seeing is a complete failure of the memory size auto-detect to read any memory whatsoever. It defaults to 4GB. Also you are missing the - before device, so I don't think it will use the algo=2 in your case above. The correct command is `pcileech.exe probe -device fpga://algo=2 -v -vv`

About the Thunderbolt, it makes sense, Microsoft added a lot of mitigations and security around Thunderbolt to protect against my PCILeech. Their OS security chief even have a nice talk about it from the Ignite conference last year.

I'm in the midst of a major rewrite of this functionality though; I plan to release it in August/September. Hopefully this will resolve your issues; or maybe not.

Until then you can try on a freshly attached device / booted system to use the -max parameter (this will also disable autodetect). Example: `pcileech.exe display -min 0x100000000 -max 0x1000001000 -device fpga://algo=2 -v -vv`

involuntarysoul, 2020-07-16:
Thank you so much for your time, I'll close this and wait for your next update

closed 2020-07-16 by involuntarysoul