ufrisk/pcileech-fpga #65: Screamer M.2 USB-C to M.2 key A+E adapter

Mera-balou, 2020-07-12:
Hi Ulf Frisk,

I already have a Screamer M.2 USB-C and a Thunderbolt adapter ([this one](https://fr.aliexpress.com/item/32867546205.html)).
I would also like to be able to connect my Screamer to a Wifi port on a laptop (M.2 key A + E).
There are a lot of different form factors and i'm afraid of being wrong. I wanted to know if you could advise me a compatible adapter between the Screamer and and the Wifi port please.

ufrisk, 2020-07-12:
There are so many different adapters around this so I'd rather not give advise so I won't give you the wrong one.

But there are plenty of a+e to standard pcie adapters out there. Maybe some to key M as well.

Mera-balou, 2020-07-13:
Yes, there are so many different adapters and formats that I get lost. 
[This one](http://www.adt.link/product/R54SF.html) seems to be suitable.

ufrisk, 2020-07-16:
I agree, it does look very nice indeed :) Maybe I shall even get one of those myself. Thanks.

craigsblackie, 2020-07-29:
I've ordered this adaptor and the Sintech. I'll see how they compare.

craigsblackie, 2020-08-04:
So, I can confirm the Sintech adaptor works perfectly. On the Dell Precision devices, I've found that the NvME port doesn't work but the Wifi a+e port does, so definitely worth investing.

craigsblackie, 2020-08-06:
I've just received the ADT-Link R54SF and it's a lot more sturdy looking than the Sintech which uses a flimsy ribbon cable. Saying that, ribbon cables are cheap so it wouldnt be too much trouble to replace if it got damaged.

closed 2024-06-01 by ufrisk