ufrisk/pcileech-fpga #66: Pcie Link Blinking

PaXePlays, 2020-07-28:
Hi :)

I have an issue with my m2 screamer (USB-C version).
My Screamer has LD3 always on and LD1 is blinking.
When I press the reset button LD2 blinks 4 times which I guess means it's flashed correctly?

->I have tried lower ram speeds.
->Secure boot, CMS, fast boot, VT-d and Intel Virtualization Tech is disabled.
->I have tried all the adapters, m2 and even in the graphics card slot.

when running "MemProcFS.exe -device fpga://algo=2 -v -vv"
I get this "unable to retrieve required device pcie id [4,v4.6,0000]"

My mainboard is the Gigabyte Z390 Aorus Elite.

I hope that you can help me figure this out.

ufrisk, 2020-07-29:
The blinking led ld1 indicate that you don't have a proper pcie connection set up. I also see you're running the most recent v4.6 bitstream. It's flashed correctly. I have a similar problem on one of my Intel NUCs; but I have not been able to pinpoint the exact problem. It seems to be with the slot sometimes requiring all 4 PCIe lanes to be set up (I currently use only one lane).

Can you please try to insert it in an ordinary PCIe slot (not the M2 slots) with the adapter LambdaConcept provides? Even an x1 slot with the smaller adapter to see if it start working?

PaXePlays, 2020-07-29:
Thanks for the reply.

It curretnly is in the x1 adapter from lambda and I have also tried the x4 adapter and m2 slot.
The screamer always behaves the same, even in the full x16 slot where my graphics card normally is.

PaXePlays, 2020-07-29:
I have tried every possible adapter on all of my pcie slots.

ufrisk, 2020-07-29:
Is the LD1 always blinking in all those setups? Or does it stay lit sometimes?

PaXePlays, 2020-07-29:
Yes LD1 is blinking in all those setups.
The only difference I noticed is that sometimes when I press the reset button LD1 turns off comnpletely until I reboot.

ufrisk, 2020-07-29:
the meaning of this is that the fpga is receiving the clock signal for the pci express, but for some reason it's not able to set up a pci express connection for some reason.

Right now I have no explanation for this. It should really work. sometimes people have reported success in disabling xmp and/or downclocking the memory and/or upgrading the bios if there is a more recent version. but somehow it feels like this may be a more fundamental issue since it's not able to get the PCIe to work at all.

closed 2020-12-12 by PaXeGAS