ufrisk/pcileech-fpga #67: ScreamerM2 self-disconnect

xNarkon, 2020-08-09:
Hey @ufrisk!

I have noticed another bug. Probably firmware related, but I'm not sure. 

Sometimes the device is disconnecting from the attacker PC, without any error - just disconnecting. After that, I cannot initialize it with VMMDLL_Initialize until I restart the target PC. Do you have any idea what is happening?

ufrisk, 2020-08-09:
I have noticed disconnects yes, you hear the sound. But for me it always reconnects. The initialization fail is a bit strange.

I'll look into the disconnects though. They bother me, but I have been unable to locate the source of them.

xNarkon, 2020-08-19:
@ufrisk how things going? Any progress?

ufrisk, 2024-01-17:
Closing issue due to old age. I believe it may be better now with recent bug fixes, if not chances are it may be something hardware related which I cannot do anything about.