ufrisk/pcileech-fpga #68: ScreamerM2 stops responding after app crash

xNarkon, 2020-08-09:
Hey @ufrisk!

Another bug, potentially with firmware. After the application which is using VMMDLL crashed without properly closing the connection to the FPGA device there is no way to connect to it until restart the target PC. All of the 3 LEDs are constantly lightning. Is there any way to restart the FPGA device via software without manually restarting the PC/Device? The strange thing is that the middle LED is constantly lightning as if the FPGA is trying to send some data, but the connection is no longer there and a new one cannot be established. Also, there is no way to close the target PC via Windows->Power off. I need to shut it down using the power button.

ufrisk, 2020-08-09:
if running the most recent bitstream version (v4.6) you may try to force a bitstream reload on the device from the usb side. You may do this by specifying devreload=1 in the device parameter, example `-device fpga://devreload=1` to the initialization parameters. You may also try to replug the usb cable.

The LED indicators are for INIT, PCIe Link, and USB-data exists in buffer (i.e. not strictly data in transit) - hence the constant light after the crash.

Why did it crash, are you suspecting leechcore.dll / vmm.dll for this; or do you believe it was due to something else?

---

Also, if you find PCILeech / MemProcFS useful please consider sponsoring the project here on Github. I see people purchasing hardware for hundreds of dollars (of which I receive absolutely zero dollars for - since I'm not related to hardware sales) just to be able to run my free open source software. Sponsorships go for as little as $2 and Github is matching it - a $2 sponsorship for you is a $4 sponsorship for me. Thank You :sparkling_heart:

xNarkon, 2020-08-09:
Crashes are only my fault. At least I haven't encountered any problems of this kind on the leechcore / vmm side yet. I will try with your proposal and will let you know soon. Cable re-plug does nothing, unfortunately :(

Thanks for your help.

Regarding sponsorship, I've wanted to do this for a while, but it wasn't available yet. Mega respects your job, you saved me a lot of time.

ufrisk, 2020-08-09:
Huge Thanks for the sponsorship 👍 

And about the issues; I know it's not perfect; I really appreciate the issue reports so I'll get a chance to actually fix the things (or at least so I hope). Some issues, like this one, is very tricky though; I have tried to locate the error but I have not been successful. I'll do another major investigation into the matter in a couple weeks time from now. I hope it's useful as-is meanwhile though. And once again, Thank You for your sponsorship 💖

xNarkon, 2020-08-09:
It's nothing in the scale of what you do. As soon as I have the opportunity, I will increase the sponsorship amount :)

If I can help you somehow, let me know.

ufrisk, 2020-08-09:
Thanks :) Keep reporting issues and bugs are actually quite awesome. Of a few I know already but I haven't been able to pinpoint the issue quite yet, of others I know nothing about until reported.

ufrisk, 2024-01-17:
I'm closing the issue due to old age.

Recently I also fixed a lot of smaller bugs in the fpga bitstreams so chances are it will be a lot more stable now.

closed 2024-01-17 by ufrisk