ufrisk/pcileech-fpga #71: Flashing Screamer M2 with custom Vendor & Device ID still shows default ID's in Device Manager?

QuiNz0r, 2020-08-27:
Hi Ulf,

thanks for your awesome work I followed your guides and project very closely. Yet I have a strange issue I cannot resolve.

I followed your guide to change Device & Vendor ID and succesfully flash my own pcileech_screamer_m2_top.bin build with openocd.
[image]
the output is just as expected so the flashing shouldn't be the problem.

However in Windows after I reboot the target Computer it stills shows the old values in Device Manager:
[image]

Any idea what could cause this? I changed Vendor ID, Device ID, serial number and enabled custom configuration space just as described in your guide.

Any help would be greatly appreciated.

QuiNz0r, 2020-08-27:
I just noticed the openocd outputs differ slightly (**DEPRECATED** messages), might that be the problem?

[image]

`Open On-Chip Debugger 0.10.0+dev-01293-g7c88e76a-dirty (2020-07-02-19:28)
Licensed under GNU GPL v2
For bug reports, read
    http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : sector 0 took 113 ms
Info : sector 1 took 108 ms
Info : sector 2 took 113 ms
Info : sector 3 took 116 ms
Info : sector 4 took 125 ms
Info : sector 5 took 114 ms
Info : sector 6 took 110 ms
Info : sector 7 took 98 ms
Info : sector 8 took 122 ms
Info : sector 9 took 118 ms
Info : sector 10 took 117 ms
Info : sector 11 took 114 ms
Info : sector 12 took 108 ms
Info : sector 13 took 102 ms
Info : sector 14 took 110 ms
Info : sector 15 took 92 ms
Info : sector 16 took 139 ms
Info : sector 17 took 105 ms
Info : sector 18 took 102 ms
Info : sector 19 took 104 ms
Info : sector 20 took 102 ms
Info : sector 21 took 104 ms
Info : sector 22 took 110 ms
Info : sector 23 took 112 ms
Info : sector 24 took 116 ms
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
shutdown command invoked`

QuiNz0r, 2020-08-27:
I was able to solve the issue by building the binary through the application and not by command line. :)

closed 2020-08-27 by QuiNz0r

ufrisk, 2020-08-27:
I'm happy the issue resolved itself. I wish you the best of luck with your future DMA explorations 👍 

---

Also, if you should find PCILeech / MemProcFS useful please consider sponsoring the project here on Github. I see people purchasing hardware for hundreds of dollars (of which I receive absolutely zero dollars of) just to be able to run my free open source software. Sponsorships go for as little as $2 and Github is matching it - a $2 sponsorship for you is a $4 sponsorship for me. Thank You 💖