ufrisk/pcileech-fpga #73: Process List Refresh loop

vinnyNC, 2020-09-06:
Using Screamer R04 with custom device ID/vendor ID. Getting process refresh loop that just runs forever. I've attached the test output. I'm running Windows 10 v2004 if that matters.

I'm curious if I just messed something up somewhere, really looking for some sort of direction on this.
[message(1).txt](https://github.com/ufrisk/pcileech-fpga/files/5179755/message.1.txt)

ufrisk, 2020-09-06:
hrm, the process refresh loop is mean to be running in the background with some periodicity in a neverending loop. this is done to refresh process lists in a convenient that minimizes delays for the user in case new processes are started and such.

now it's super hard for me to see with what periodicity this occurs; if it takes some time between refreshes; 5s is default for ProcessRefresh Partial and 15s for ProcessRefresh Total. Is this the behavior you're seeing? Or is it happening very quickly for you?

If you wish to disable refreshes you may start MemProcFS / vmm.dll with extra command line argument `-norefresh` - but memory structures will start to drift after a while so this is not really recommended if running over a longer period. You may also configure with some options the periodicity of these checks.

----

Also super thanks for the sponsorship. Not many people realize that I'm not affiliated with hardware sales and that I get nothing out of it. Thank You!

vinnyNC, 2020-09-06:
I didn't get the loop with default firmware, so any idea why this would continue in the foreground and not allow the test to end?

The command I'm running is MemProcFS -device fpga -v -vv

ufrisk, 2020-09-06:
That debug output happens when you run with the `-vv` (extra verbose) option. mbe you didn't start MemProcFS with the -vv option the first time? It should not be related to which fpga firmware you're running.

Anyway, that is expected output when running with `-vv` if it refreshes every 5 seconds.

vinnyNC, 2020-09-06:
I know the command I ran hasn't changed, so I'm kinda confused. The only difference is I'm using win 10 2004 instead of 1909. I'll try downgrading.

vinnyNC, 2020-09-07:
Well, a dumb error on my part. It kept looping because I had dokany installed while I previously didn't. It wasn't even something I thought about until I downgraded, but at least I know now.

Thanks 👍

ufrisk, 2020-09-07:
Awesome that it resolved itself :) I wish you the best in your future DMA explorations. Please let me know if you should run into anything again or if you're missing features. I'm closing the issue since it's now resolved.

closed 2020-09-07 by ufrisk