ufrisk/pcileech-fpga #74: where do i edit config space on ISE 14.7 on Unix ?

mmollet, 2020-09-08:
Hi, ive made alot of custom firmwares for ScreamerM2, now i tried with an SP605 but i cant find the config space file to edit it like the this one - pcileech_cfgspace.coe

ufrisk, 2020-09-11:
Thank you for the contribution to the project :sparkling_heart:

About the SP605, I have not added the custom configuration space capability to it. It was quite some time since I looked into that bitstream design. I would need to install the dev environment and look into how much work it would be to support it, and some other features of more recent designs, if feasible at all. I have a few things to prepare for next week so unfortunately this may be some weeks away. I'll keep you updated.

amdeveloping, 2021-03-26:
Would love to know if this is still being looked into

ufrisk, 2021-03-26:
Thanks for the reminder. I remember looking into it but I gave up at the dev env installation since it required virtualization enabled and nested virtualization is not yet supported for my cpu last time.

Anyway; I just installed an old copy of Windows7 and the dev environment to have a look. It should be possible to do this. The Xilinx PCIe core itself supports it and there are enough free resources to implement this.

But there is still quite some work involved for me to get it working; especially with testing and trouble shooting any errors. I wouldn't be surprised if it would end up being a couple of days work in total. I just don't know if there are enough people that would be interested in this; except for you two; to warrant me to put my time into this instead of other aspects around my project.

I'll keep it on my todo list for sure (thanks for the reminder) and I'll look into it when/if I find the time; but it may take some time.

ufrisk, 2024-01-17:
I'm closing this issue due to old age. Unfortunately I don't forsee that I'll re-visit the SP605 and upgrade the SP605 firmware. It will still continue to work as-is though.

closed 2024-01-17 by ufrisk