ufrisk/pcileech-fpga #75: Thank you

QuiNz0r, 2020-09-16:
Hi Ulf,

no issues here, just wanted to say thank you for this amazing project I've been following and studying since months. Everything works flawlessly on a ryzen 3900xt. I even got pcileech to work in unity3d to visualize data thanks to your new c# api.
I just donated and I urge others to do so too so you can keep up with your work.

one question: any progress on usermode dll injections?

Have a nice day and I'm looking forward to future developments, you teached me a ton.

ufrisk, 2020-09-17:
And thank you so very much for contributing to the project 💖 

It's awesome to see that the C# API is useful. It's very new and I think you may be one of the first users of it (it's hard for me to know though). If you run into something you feel is missing with it and/or if you find something strange (bugs?) please let me know.

About usermode injections; I assume you wish to do it without having a pcileech kernel module inserted into the system first? I haven't looked into that for a while now. The tricky part for me is to gain code execution in a consistent way across multiple types of user mode processes. Since I can't call API's (I'm only able to read/write memory) I need to place a hook somewhere. And not all user mode processes call all types of code so it's a bit tricky. I have a few other things I'd like to focus on first; but this is definitely an area I'll look into again - even tho it may still be some time away.

QuiNz0r, 2020-09-17:
Other than an occasional hiccup when reading memory continuously it's been working absolutely smooth and flawless 👍
I imagined it's very complex without a kernel module on the target system so I wouldn't focus on it either, just an exciting Idea

I'm currently initializing with the parameters -device fpga (obviously) and -memmap auto. Anything I could try to reduce hiccups? (might even be hardware related) feels like the transfer rate bottlenecks for a second and goes back to normal very quickly

Thanks again ❤️

ufrisk, 2020-09-17:
I'm aware of the hiccup issue; I just have no clue why it's happening and it's happening quite randomly as well. I have no solution for it currently and I have been unable to find what may be causing this issue so far - even though I'll keep looking.

I noticed that connecting a USB hub in-between removes the issue for me; but I have had reports from others saying that it's not doing anything for them.

Hopefully I'll locate the issue; but chances are it will be there for quite some time :/

QuiNz0r, 2020-09-17:
oh that's interesting I'll try a hub then! 
So you just connect the hub to the attacker PC and then go from the hub to the target pc via usb? do you use a powered hub or passive?

ufrisk, 2020-09-17:
I used a powered one; but as I mentioned others haven't been as successful in me with regards to that so it's something that may be worth a try, but I wouldn't count on it. Please let me know how it goes.

QuiNz0r, 2020-09-23:
just a quick report, I purchased a new powered usb3 hub and couldn't get pcileech to work for some reason, it initialized fine but I wasn't able to read any data. most likely an issue with the hub. Nonetheless pcileech is working fine even with some occasional stutter for me

ufrisk, 2020-09-27:
An update from my side as well, I have looked into the connection hiccups and I'm unable to pinpoint what it may be due to unfortunately; chances are it will be around for some time unfortunately since I'm at a total loss at this moment what this may be due to.

ufrisk, 2020-10-11:
Another source for more minor hiccups may be the background refreshes that run with some periodicity to keep things like process list and process memory allocations up to date. You may disable this with the command line option `-norefresh`.

imerzan, 2022-02-18:
Did some testing and it does indeed seem to be the periodic refresh, because -norefresh helped a TON (although did pose other problems).

I found out that my USB3 cable was faulty and only operating at USB2 speeds (~24mb/s). I now have a brand new USB3 cable and its running at ~130mb/s and this issue has pretty much evaporated.

Just wanted to provide an update that I no longer have an issue here. Thank you!!

closed 2024-01-17 by ufrisk