ufrisk/pcileech-fpga #76: Minimum FPGA hardware requirements label:question

Vensis, 2020-10-11:
Hi,

this is just a question. 
I was wondering what would be the minimum hardware requirements to succesfully flash pcileech to some simple FPGA board and make it work. 
Do you mind sharing the theoretical minimal set of conditions that has to be met to run flash and use pcileech?

Great project btw.

Regards,
Daniel

ufrisk, 2020-10-11:
I support select hardware devices.

Each device requires some customization of the firmware. Due to this it's not really possible for me to set a minimum requirement. It's all up to person doing the customization about what tradeoffs they are willing to make, i.e. how much performance impact is acceptable for a weaker device.

Doing customizations to other boards than very similar to the ones supported is not a small job though. If interested it's probably easier to get one of the supported devices than spending lots of hours for something that may or may not work.

Vensis, 2020-10-12:
Of course I am aware that using board different from those which you support requires some effort, for instance in properly mapping I/O.

Nevertheless, what do you think would be the weakest FPGA to run pcileech, in terms of FPGA minimum architecture family that might for some reason be required to sucessfully flash pcileech?

How many logic gates are required to be able to flash pcileech?

What is minimum set of additional interfaces talking to FPGA required to start using pcileech?
PCIe connection to memory is obvious, I assume that I would also need some USB interface, preferably 3.0 and some fast memory, is that correct? What would be the required size of DDR3 to start using this tool?

Vensis, 2020-10-13:
Do you mind answering?

ufrisk, 2020-10-13:
I don't have any numbers, sorry about that, you can check it out if you build my projects in Vivado though.

Generally it's a really simple design, it basically about shuffling data, with minor processing, between a FT601 chip for the USB3 and a PCIe core. TLP generation and such happens on the software side and raw TLPs are sent over the USB3.

FT601 is 32-bit and 66/100MHz.

Almost all logic in my designs are used as memory buffers (FIFOs); but you may cut these down if you accept lower performance, or as you mention use external DDR which I currently do not make use of.

If you do wish to use something else than a FT601 USB3 (or custom UDP/IP protocol) chances are that you would need to make modifications to the PCILeech/LeechCore software itself.

closed 2024-06-01 by ufrisk