ufrisk/pcileech-fpga #78: How to make the coe file of PCIe configuration space effective.

black-pigeon, 2020-11-27:
Hi bro,
I followed the build.md, and modified the coe file of screamer_m2 , when I flashed the .bin file,  the device has changed it's type, but when I used  `lspci -d   -xxx` to confirm the PCIe configuration, I get the hexadecimal dump of the whole PCI configuration space, which is no the same as the .coe file.
My goal is to make the screamer_m2 to be a sound card or samething.
Thanks!

ufrisk, 2020-11-27:
Hello,

if you do `lspci -d -xxxx` you should see your new custom config space in the upper bytes (above 0x100) and also some bytes below 0x100. If you are seeing all zeroes there please change [this bit](https://github.com/ufrisk/pcileech-fpga/blob/9b61d7cb1d7eab76dec6de79dbda0b3061476718/ScreamerM2/src/pcileech_fifo.sv#L264) to enable custom config space.

Also master abort bit auto-clear is disabled by default; you may enable it by changing [this bit](https://github.com/ufrisk/pcileech-fpga/blob/9b61d7cb1d7eab76dec6de79dbda0b3061476718/ScreamerM2/src/pcileech_pcie_cfg_a7.sv#L207)

config space required for the Xilinx PCIe core to work always takes priority over custom config space though; If you wish to change the majority of the bits below 0x100 you'll have to look up what they stand for and alter the values in the Xilinx Vivado PCIe wizard by double clicking on the PCIe core. This may be a long and tedious iterative process to get it right.

I'm not able to give more pointers than this and also I won't be able to help you create such a config space; but I wish you good luck in your PCIe DMA research.

black-pigeon, 2020-11-28:
Thanks  for your reply, this is an excellent project. By the way, I am a FPGA developer familiar with Verilog, but your project is built with SystemVerilog. Can you recommend some learning resources of SystemVerilog, I will be very grateful.

ufrisk, 2020-11-28:
I used only a few systemverilog constructs; mainly "interface" which allows me to bundle wires. Other than that I think most things should be more or less plain verilog.

and no; I can't recommend any good reading; I learnt my very basic systemverilog from google more or less :\

closed 2020-12-31 by ufrisk