ufrisk/pcileech-fpga #82: FPGA: ERROR: Unable to connect to USB/FT601 device

ryzen7i7, 2021-01-14:
I'm using LenovoX13 for attack, with R04 screamer m.2.

I used openocd on windows for flashing with no error(same output as lambda concept documents).
I tried pre-flashed gateware and then followed lambda concept and pcileech-fpga install guide, but I get below error.
  pcileech.exe dump -out none -v -vv -device fpga
   DEVICE: FPGA: ERROR: Unable to connect to USB/FT601 device [0,v0.0,0000]
   PCILEECH: Failed to connect to the device
I have searched issues, is this correct with the error that USB is not connected?
Attack PC have FTD3XX.dll on windows.
I use a type-c to type-a adapter and also connect directly to type-c port(which supports data transfer) and get same output.

Then R04 is (probably) successfully detected on target pc as:
 Other devices > Ethernet controller
  Status: 
   The drivers for this device are not installed(Code 28)
  Events:
   Device PCI \VEN_10EE&DEV_0666&SUBSYS_000710EE&REV_02\6&9f5003c&0&0020020B was configured.
 and icon have yellow exclamation point.

Are there any other errors or info that we need to resolve?
thank you.

ufrisk, 2021-01-14:
The device seems to be correctly flashed. This in combination with the error message indicates you have a problem with your USB connection to the device.

Please verify that the USB-device is detected on your target PC; it should show up as a USB controller.

If it's not showing up or if there is an error make sure your computer have an internet connection so that windows may download the driver from windows update. Also try different usb ports, cables, etc.

[image]

ryzen7i7, 2021-01-14:
[image]
How should i do if shown like this?

ryzen7i7, 2021-01-14:
P.S. I modified vendor id and device id with vivado then built and flashed it, but nothing has changed. No error output by openocd

ufrisk, 2021-01-14:
It looks about correct, the usb-c variant looks a bit different from the usb3 one I have (I have misplaced my usb-c device and I'm not too keen on purchasing a new one).

So if I understand you correctly this device was working, you reflashed it and it stopped working?

Please try to reflash it with the default bitstream I have on my github page and see if it starts working again. Maybe you accidentally changed something else. If that isn't working there may be a hardware issue.

ryzen7i7, 2021-01-14:
I tried it after receive it then didn't works as well, I searched Issues and saw need to flash a new one, so I reflashed it.

And I'm currently building one downloaded from release page, Sorry if I missed that, but is there a precompiled bitstream?

ufrisk, 2021-01-14:
If it never worked mbe it's a DOA? It's supposed to be pre-flashed with a default bitstream.

ryzen7i7, 2021-01-14:
I flashed then get this output:
C:\Users\ryzen\Desktop\openocd\bin>openocd.exe -f flash_screamer.cfg
Open On-Chip Debugger 0.10.0+dev-01293-g7c88e76a-dirty (2020-07-02-19:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
DEPRECATED! use 'adapter speed' not 'adapter_khz'
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : JTAG tap: xc7.tap tap/device found: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
Info : sector 0 took 103 ms
Info : sector 1 took 96 ms
Info : sector 2 took 115 ms
Info : sector 3 took 101 ms
Info : sector 4 took 116 ms
Info : sector 5 took 132 ms
Info : sector 6 took 103 ms
Info : sector 7 took 98 ms
Info : sector 8 took 117 ms
Info : sector 9 took 116 ms
Info : sector 10 took 125 ms
Info : sector 11 took 119 ms
Info : sector 12 took 119 ms
Info : sector 13 took 103 ms
Info : sector 14 took 104 ms
Info : sector 15 took 88 ms
Info : sector 16 took 104 ms
Info : sector 17 took 104 ms
Info : sector 18 took 105 ms
Info : sector 19 took 109 ms
Info : sector 20 took 106 ms
Info : sector 21 took 108 ms
Info : sector 22 took 106 ms
Info : sector 23 took 92 ms
Info : sector 24 took 107 ms
Info : Found flash device 'issi is25lp256d' (ID 0x0019609d)
Warn : device needs paging or 4-byte addresses - not implemented
shutdown command invoked

DEPRECATED! use 'adapter driver' not 'interface' is worrying me, but this is normal, right?

Then I checked devices in victim PC, I see Other Devices>Ethernet Controller.
I'm not well versed in FPGA devices, but it doesn't show up on DOA(or any damaged), does it?

ufrisk, 2021-01-14:
I have no clue about OpenOCD.

But; the device is clearly working on the PCIe side. The USB shows up in the device manager (i.e. cable should be ok I'm guessing; but try another just to be sure also in another USB port or ideally in another computer); but yet pcileech on your computer is unable to contact the FPGA. It's not a driver issue since the message should be something else.

It's a really strange error; maybe you got a bad device. check the soldering around the ftdi chip to see if it looks ok or not.

ryzen7i7, 2021-01-14:
Sorry for bad photo, Is it normal that there are no resistors attached to upper left like R39, R40 silks.
[image]

ufrisk, 2021-01-14:
it's the same on my device

ryzen7i7, 2021-01-14:
there are no problems with soldering around ftdi chip.

ufrisk, 2021-01-14:
Then I just don't know. it's a rather strange error. I double checked my latest releases and they are working fine; you're also the first one that reported issues around this.

Please know that I provide this software as a hobby project. I make very little money out of this. other companies like LambdaConcept are selling products; of which I get absolutely nothing out of their sales.

With this being said I just can't provide endless hours of unpaid support just so others can make money off my free open source hobby projects. I regret that I'll be closing this issue with the recommendation that you contact LambdaConcept about your issue for further support. I wish you the best and I hope you'll be able to resolve this issue. Good Luck and best wishes.

closed 2021-01-14 by ufrisk

fisherwise, 2021-01-19:
sorry to re-open this. Run into this issue today and switching a usb-c cable fixes the issue.
Not sure what's the underlying issue was, as the old cable works on other secnarios (connecting a monitor etc.)

ufrisk, 2021-01-20:
Awesome that you got it to work after you tried another cable :) Unfortunately there is nothing I can do about bad cables, glitchy connections and things like that. It's super nice that you got it to work though and that you let me know about it. I wish you the best in your future DMA investigations :)

---

Also, if you should find PCILeech / MemProcFS useful please consider [sponsoring](https://github.com/sponsors/ufrisk) the project here on Github. I see people purchasing hardware for hundreds of dollars (of which I receive absolutely zero dollars of) just to be able to run my free open source software. Sponsorships go for as little as $2 and Github is matching it - a $2 sponsorship for you is a $4 sponsorship for me. Thank You 💖

Edward-P-Astbury, 2023-10-22:
Sorry to comment on this closed issue, but after tedious troubleshooting I resolved the issue by uninstalling the 'FTDI SuperSpeed-FIFO Bridge' driver from within device manager.