ufrisk/pcileech-fpga #83: Screamer M.2 USB-C not recognized by host pc, but usb3 adapter works

fisherwise, 2021-01-15:
Posting here in hope that someone else had this issue/resolved it.

I plugged the usbc cable into my host and it's not recognized in device manager, but when I plug it it with usb3 adapter it works.
Also If I'm plugging the usbc into victim pc it's recognized.

I've no clue what's happenning.

alexanderzinolidis, 2022-08-17:
I have this exact same issue and it's driving me crazy.

Host is a MacBook pro running Windows on Bootcamp. Usb-c directly into the Mac doesn't work. Using one of those small USB-C to USB3 adapters works, but I guess this limits performance pretty substantially?

ufrisk, 2022-08-17:
@alexanderzinolidis what performance in MB/s are you looking at when doing `pcileech.exe dump -out none`?

alexanderzinolidis, 2022-08-17:
Around 85 MB/s

ufrisk, 2022-08-17:
I don't know. The FTDI USB chip is only USB 3.0. It might have something to do with that. But it should be handled transparently.

85MB/s is not too bad but it's on the slow side of things, it should be around 120MB/s usually.

I don't know what the issue is, but I'm pretty sure it's hardware/driver related somehow and that there isn't much I can do about it on the software side of things. It's more an issue for LambdaConcept to take a look at (if using the Screamer boards).

alexanderzinolidis, 2022-08-17:
Yeah, I kinda figured this is completely unrelated to pcileech. I will try to dump directly from the target PC which has a usb-c port that works with the chip. That machine is way beefier so should at least confirm that the issue could be the Macbook.

Thank you so much for your time and the awesome work you have done with this project.

alexanderzinolidis, 2022-08-17:
It is indeed the MacBook or Bootcamp that is causing the poor performance. When dumping directly from the target pc using the same cable and adapter I am maxing out the squirrel at 150 MB/s

Could also be the usb hub that I have between the chip and the Mac..

ufrisk, 2022-08-19:
USB hubs often adds latency and will slow down things. Somethings substantially. Please try without the hub and you might get higher speeds. Please let me know what kinds of speeds you'll get without the USB hub (if you're able to try it).

I'm closing this issue since there isn't really anything I can do about this and 85MB/s while not great is not terrible either.

closed 2022-08-19 by ufrisk